rusefi/rusefi #3684: rename hellen72 paths

andreika-git, 2021-12-09:
We are adding more pins to hellen72, like in other Hellen-Miata boards. 
Maybe we should change the board-related paths:

1) Rename the repo
Now: https://github.com/rusefi/hellen72/
Proposal: https://github.com/rusefi/hellen-NB2/

2) Rename the wiki
Now: https://github.com/rusefi/rusefi/wiki/Hellen72
Proposal: https://github.com/rusefi/rusefi/wiki/Hellen-NB2

3) Keep the old short /s/ link (printed on the old boards) and change it to the new redirect:
  https://rusefi.com/s/hellen72   ---> https://github.com/rusefi/rusefi/wiki/Hellen-NB2

4) Add a new short /s/ link:
  https://rusefi.com/s/hellenNB2  ---> https://github.com/rusefi/rusefi/wiki/Hellen-NB2

rusefillc, 2021-12-09:
Step 1 done @BeerMoneyMotorsports fyi

BeerMoneyMotorsports, 2021-12-10:
Excellent, you have beat me to the request @andreika-git

rusefillc, 2021-12-10:
2,3,4 done

do we want to rename the binary? feel free to PR binary rename or poke me. If we rename binary we would have to change link in two steps - snapshot link now and release link once we do next release.

BeerMoneyMotorsports, 2021-12-14:
Binary can stay the same for now.  i do not see a reason to change it?

closed 2021-12-14 by rusefillc