rusefi/rusefi #3701: Need better way to Enrich on low cylinder engines

NMSTEC, 2021-12-20:
Request: AlphaN table override VE for idle ~ 2000Rpm. Table.

https://rusefi.com/online/view.php?log=487

`Matthew Kennedy  11:18 AM
I think we need better strategy, not better calibration`

Due to low amount of cylinders, MAP based fuel maps are too slow to react, and cause delay in fueling. We set TPS vs RPM as VE, and reaction is better, but still too slow. Artificially increasing VE in idle areas helped, but still sucks.
[image]

rusefillc, 2021-12-21:
i wonder if that should be lua as well, i.e. good excuse to try fully Lua fuel model

blundar, 2022-01-02:
The title of this ticket reflects NMSTEC seeking a way to make his engine
respond to input from the driver faster, i.e. improve transient behavior.

I think that there is more to this ticket than just that.  Each method of
predicting airflow (and also fuel) entering the engine has its strengths
and weaknesses.  Each method also has conditions in which it doesn't work.

Being able to switch between metering methods, or perhaps even blend
between them is a positive feature beyond the impact on transient fueling
that brought this up.

NMSTEC, 2022-01-13:
There's also a work around way of doing this, although not proper is used by some manufacturers. We could delay the opening of the ETB. Request vs actual. Open it just enough for fueling to catch up, then let ETB go back to being controlled by APP request.