rusefi/rusefi #3717: Miata triggers review

BeerMoneyMotorsports, 2021-12-23:
Problem- Miatas suck to crank up and no level of tuning seems to really help.  Ive personally spent days trying to get cars to crank reliably.  

Miata triggers seem to struggle during cranking.  I personally see a trigger error every time i go to start. @zhoob2004 also sees this issue.  This is something i see from both NA CAS and NB stock crank/cam triggers.  Below is a link to a log from @zhoob2004 on the 96 with proteus.  I will be updating with logs from nb1 in the coming days.  

To note, by changing from stock NB triggers to 36-1 crank trigger, the problem is resolved.  My thoughts are we are missing something on the triggers OR the 36-1 hides terrible cranking tuning.  

https://rusefi.com/online/view.php?log=515

mck1117, 2021-12-23:
@zhoob2004 or @BeerMoneyMotorsports - can you try with these gap override settings?  These are derived from data in Zach's log (515).

[image]

BeerMoneyMotorsports, 2021-12-27:
@mck1117 nb1 recording.  4 starts.  3rd was no start while 4th fired right up.  they were sequential attempts. no time between.

https://rusefi.com/online/view.php?log=517

rusefillc, 2021-12-27:
@BeerMoneyMotorsports do you have  logic analyzer?

mck1117, 2021-12-27:
We've been chatting on slack, probably don't need a logic analyzer with the recently improved log fields

BeerMoneyMotorsports, 2021-12-27:
@rusefillc i do, i can implement it however directed.

mck1117, 2021-12-27:
Let's use this issue for NA, and #3721 for NB

mck1117, 2021-12-27:
@Zhoob2004 and @BeerMoneyMotorsports will report back here once they've tried those adjusted sync settings for NA

BeerMoneyMotorsports, 2022-01-01:
lemon with latest snapshot and no modification to tune.  76f ambient cold start.

https://rusefi.com/online/view.php?log=529

BeerMoneyMotorsports, 2022-01-01:
log with trigger gap override settings

https://rusefi.com/online/view.php?log=530

[image]

BeerMoneyMotorsports, 2022-01-01:
These settings work fantastic.  Solid improvement. Consistent and fast startup.

**Please keep ticket open for further cold weather tests**

[image]

BeerMoneyMotorsports, 2022-01-03:
near freezing start.  did not catch first try

https://rusefi.com/online/view.php?log=542

BeerMoneyMotorsports, 2022-01-07:
@mck1117 the new settings seem to have done great.  i think this ticket can be closed now?

mck1117, 2022-01-07:
Sure, if we're happy with NA sync, I'm happy to close.

closed 2022-01-07 by mck1117