rusefi/rusefi #3718: Proteus H7 red LED no USB

rusefillc, 2021-12-23:
master H7 Proteus has RED led, blinking blue and no USB comms

I've broken H7 build somewhere around https://github.com/rusefi/rusefi/commit/15daec77e6f99e94735edaa7aaf2807e8c79552f

looks like
a) CAN2 is still not properly enabled on H7
b) it fatals in a way which does not get us a working USB which is pretty desired even for FATAL

mck1117, 2021-12-24:
The H7 CAN driver doesn't support dual-CAN yet, afaik.

rusefillc, 2021-12-24:
Ohhh

let me make CAN default pinout sensitive of Fx vs H7

rusefillc, 2022-01-15:
@NMSTEC please notice this ticket. I am not aware of level of effort so I assume the worst, i.e. I assume big effort. I know you are interested in dual CAN bus please note that this is impossible on H7 as of today.

mck1117, 2022-01-16:
It's very doable.  It'd take me a few hours with a live H7 sitting in front of me. If it becomes a priority I can make it a priority.

rusefillc, 2022-01-16:
Not a priority since we only have 25 pieces in reserve