rusefi/rusefi #3720: stm32_stop just reboots stm32f4discovery

rusefillc, 2021-12-26:
stm32f4discovery with fresh firmware
PA0 jumper wire to GND, no other wires
mini USB, micro USB

send stm32_stop via console

expected: some sort of stop
actual result: firmware reboots

rusefillc, 2021-12-29:
@NMSTEC do you have an stm32f407 discovery board?

NMSTEC, 2021-12-29:
I do not. Will order from the link you sent, and will fuck with proteus upon my return.

rusefillc, 2022-01-03:
@NMSTEC report same or similar on Proteus F7 with some hw mod

NMSTEC, 2022-01-03:
STM32F7 on proteus. Cut PA0 trace, soldered on jumper to RX pin of CAN. Can is not active at this time, and running on bench from USB only. stm32_stop forces reboot. Cut jumper, leaving PA0 hanging in the air, problem remains.

rusefillc, 2022-01-06:
maybe relevant:
```
The important part is to always clear the wakeup flag (via PWR_CR_CWUF_Msk) in the PWR->CR register. Otherwise the device will immediately wake up after attempting to enter the stop mode.
```

NMSTEC, 2022-01-12:
Resolved: https://github.com/rusefi/rusefi/pull/3783

mck1117, 2022-01-16:
dddddone!

closed 2022-01-16 by mck1117