rusefi/rusefi #3721: NB1/NB2 triggers review

mck1117, 2021-12-27:
Breaking off NB1/2 as its own issue, as it's a different decoder to NA.  #3717 

@BeerMoneyMotorsports says nb1 recording. 4 starts. 3rd was no start while 4th fired right up. they were sequential attempts. no time between.

https://rusefi.com/online/view.php?log=517

mck1117, 2021-12-27:
In that log:

- Crank looks great.  We sync it no problem.
- VVT makes a bunch of garbage right at the start of cranking, what's up with that?

[image]

That mess of a blob was a total of 33 VVT edges in 0.32 second.

mck1117, 2021-12-27:
Third attempt in that log is an even worse disaster:

[image]

mck1117, 2021-12-27:
And the 4th attempt is perfect!?

[image]

BeerMoneyMotorsports, 2021-12-27:
log of second NB1.  had some oddities of rpm spike.

https://rusefi.com/online/view.php?log=518

rusefillc, 2021-12-27:
so you need three points to connect logic analyzer

GND - we usually have it on proto area
for CAM/CRANK the simplest option could be to solder little wire stubs at the back side of the PCB right to relevant ECU connector pins

0.5M is good enough rate for our purposes while saving logic analyzer trace. Please IM if I can help with logic analyzer UI.

BeerMoneyMotorsports, 2021-12-27:
Second NB1 updated firmware.  
[image]

https://rusefi.com/online/view.php?log=519

BeerMoneyMotorsports, 2021-12-27:
> so you need three points to connect logic analyzer
>
> GND - we usually have it on proto area for CAM/CRANK the simplest option could be to solder little wire stubs at the back side of the PCB right to [...]

i will rig one up for testing tomorrow.  thank you.  i have used it previously on an mre but it has been a while

BeerMoneyMotorsports, 2021-12-27:
1st NB1 updated log.  no start on first few attempts.  only started on last attempt.

https://rusefi.com/online/view.php?log=520

mck1117, 2021-12-27:
520 has the same mess as previous logs above, maybe a bad sensor??

BeerMoneyMotorsports, 2021-12-27:
logs

1st NB(orange car)
517 & 520

2nd NB(silver car)
518 & 519

BeerMoneyMotorsports, 2021-12-28:
Lets ignore 1st NB logs for this ticket as it has problems.  Use only 2nd NB logs

BeerMoneyMotorsports, 2022-01-12:
nb triggers have been much improved.  this is ok to close.

closed 2022-01-12 by rusefillc