rusefi/rusefi #3724: HellenNA8_96 config folder

BeerMoneyMotorsports, 2021-12-29:
@rusefillc could you assist me by creating the files needed for the HellenNA8_96?  Then i can get started writing the yaml.  

https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen

rusefillc, 2021-12-29:
@BeerMoneyMotorsports I've just added empty main.yaml file

rusefillc, 2021-12-29:
@andreika-git what shall we do about other firmware files? which board is closes I can at least copy-paste files from another folder?

BeerMoneyMotorsports, 2021-12-29:
@rusefillc this board is basically the same exact thing as the NB1.  same chip and modules.  main difference is pinout changes.

BeerMoneyMotorsports, 2021-12-29:
> @BeerMoneyMotorsports I've just added empty main.yaml file

thank you!

BeerMoneyMotorsports, 2022-01-07:
Closing, files created and firmware tested successfully.

closed 2022-01-07 by BeerMoneyMotorsports