rusefi/rusefi #3733: TunerStudio doesn't like arrays in outputchannel

racer-coder, 2022-01-03:
At least I think that's the error.  TunerStudio says: 
2 Errors:
Error: Error: Malformed [OutputChannel] entry, Problem at:
[mainController.ini]:[Line:2105]: alignmentFill_at_338 = array, U08, 338, [2], "units", 1, 0

Error: Error: Malformed [OutputChannel] entry, Problem at:
[mainController.ini]:[Line:2105]: alignmentFill_at_338 = array, U08, 338, [2], "units", 1, 0

Looking at rusefi.ini, I see:
m_requested_pump = scalar, S16, 334, "deg", 0.02, 0                                                 
boostControlTarget = scalar, S16, 336, "deg", 0.03333333333333333, 0                                
alignmentFill_at_338 = array, U08, 338, [2], "units", 1, 0                                          
fuel_requested_percent = scalar, F32, 340, "v", 1, 0                                                
fuel_requested_percent_pi = scalar, F32, 344, "v", 1, 0                                             

I'm guessing it doesn't like 'array' and expects 'scalar', but I could be wrong.

rusefillc, 2022-01-03:
looks like we need a different alignment Fill strategy for output channels

as a work around explicit manual unused field should do the trick

racer-coder, 2022-01-03:
I don't know if it even matters, the offset is explicitly listed in the variable.  So I don't think you need filler variables.  You could probably just omit it altogether.

mck1117, 2022-01-03:
You need the fillers to guarantee that the C layout matches the ini.

Just add a manual fill and move on.

mck1117, 2022-06-23:
this is fixed in new parser

closed 2022-06-23 by mck1117