rusefi/rusefi #3739: GHA execution throttling

rusefillc, 2022-01-04:
In order to cut the pipeline let's execute some jobs less often than always

rusefillc, 2022-01-04:
@chuckwagoncomputing I've failed to implement this :( Any chance you would be interest in this approach to GHA performance?

chuckwagoncomputing, 2022-01-06:
This issue isn't actionable as-is. Please clarify how it should be determined whether a job should be run. Coin toss? Phase of the moon? Only on Tuesdays? The color of the Queen's hat?

rusefillc, 2022-01-06:
@chuckwagoncomputing my bad

see https://github.com/rusefi/rusefi/commit/0c22c9ffb3aee4494a9221537b984bc3419f320e which is part of master now. it shows skip-rate values and execution-throttle.sh tool which is currently unused in master.

rusefillc, 2022-04-15:
@chuckwagoncomputing does it look like we currently skip at all?

chuckwagoncomputing, 2022-04-15:
Looks like it's working to me. Here is an example: https://github.com/rusefi/rusefi/runs/5998787781?check_suite_focus=true

rusefillc, 2022-04-15:
[image]

closed 2022-04-15 by rusefillc