rusefi/rusefi #3742: LUA read ECU state: Cranking, Idling, Cruising

NMSTEC, 2022-01-04:
Problem: due to specifics of the way harley runs, I have to implement its own logic in LUA to tell BCM if we are running or cranking.

Want: I want to read if ECU is in Cranking state, idling, cruising, or accelerating state.

rusefillc, 2022-01-04:
I am not sure we even have cruising or accelerating state to expose.

With current limited description my response is "please compare current RPM to cranking RPM setting, both are already available to Lua"

mck1117, 2022-01-04:
we /do/ have bits with non-trivial calculation for:
- idling
- not-idling
- coasting (foot off pedal above idle RPM)
- cranking
- running

rusefillc, 2022-01-04:
@mck1117 as is I am assuming that the simplest possible implementation would be sufficient.

As is this ticket is asking for something too vague and too random. @NMSTEC please understand where I am coming from - I really want the simplest possible solution, for that more than just "specifics of the way harley runs" would be needed.

NMSTEC, 2022-01-04:
Ok, we had this conversation before. No, comparing cranking rpm to current is not a viable solution.

BCM would like to know when the engine is Cranking, and when its Running. If ECU tells BCM its now in running sate, BCM shuts starter off. If ECU tells BCM its cranking, BCM fucks with fuel pump pressure, and then also shuts down after certain period of cranking time.

If I enter a RPM too low, a quick cough will shut off cranking. If I enter too high, it'll shut off on its own over time. Right now im using LUA one shot code, to confirm engine is running after starter is released and engine RPMs are above 500 for a period of time. Yeah I can keep it this way, however its redundant. 

Last time we spoke about this, we confirmed that once rusEFI leaves cranking RPMs, it doesn't reenter it. So I can save plenty of lua code and time, simply at least asking hardware if its out of cranking logic. 

Most importantly I need CRANKING. The rest would be lovely to have, but non critical.

rusefillc, 2022-01-05:
@NMSTEC how does https://github.com/rusefi/rusefi/commit/c66dcaf45472ccc442dc9521a8565269320ee543 look?

NMSTEC, 2022-01-05:
State 1 is perfect, many thank you!

closed 2022-01-05 by rusefillc