rusefi/rusefi #3753: stop/standby for F4

rusefillc, 2022-01-07:
we need magic same/similar to F7 on F4

rusefillc, 2022-01-07:
there is little chance of Harley being F7. realistically the most probable platform would be F4

@NMSTEC do you have any F4 test hardware? those nucleo boards you've ordered were those F4 or F7?

mck1117, 2022-01-07:
F4 doesn't have the memory for what he's doing in Lua.

rusefillc, 2022-01-07:
40x does not have the memory or 42x does not have the memory?

NMSTEC, 2022-01-07:
F7 Preferred. I have no F4 test HW.

rusefillc, 2022-01-07:
in my opinion first patch has to be assembled by JLC which means F4
i only have 25 H7 chips total

digikey has qty 10 of NUCLEO-F429ZI please order a few ASAP

NMSTEC, 2022-01-07:
Done.

rusefillc, 2022-01-15:
@NMSTEC standby is universal F4 right?

@mck1117 @NMSTEC do we care about stop at all or is it time to remove it and move on?

NMSTEC, 2022-01-15:
Stop can be woken via CAN with no HW modifications on F7. Standby can only be waken up by dedicated pins.
F4 Can only be woken via PA0 on either way.

Standby has much smaller current consumption, stop less so.

Please leave both in, until we get TJR done. and unless we are running out of space, I'd still leave both in for a deepsleep option to prevent battery discahrge.

closed 2023-01-30 by rusefillc