rusefi/rusefi #3754: stop/standby for H7

rusefillc, 2022-01-07:
we need magic same/similar to F7 on H7

#3747

rusefillc, 2022-01-07:
there is a chance of H7 Harley

@NMSTEC can you help? mouser has NUCLEO-H723ZG in stock

mck1117, 2022-01-07:
you don't want an H723, it has small flash memory

rusefillc, 2022-01-07:
@mck1117 is STM32H723ZG https://www.st.com/en/microcontrollers-microprocessors/stm32h723zg.html with "1 MByte Flash, 564 KB RAM, 550 MHz CPU" not sufficient for rusEFI or Harley?

NMSTEC, 2022-01-07:
Please let me know what to buy and what to do.

rusefillc, 2022-01-07:
@NMSTEC please order qty 3 NUCLEO-H723ZG from mouser

@NMSTEC please attempt a clarification from @mck1117 with some technical argument better than "small"

mck1117, 2022-01-07:
@rusefillc you need to slow down

The 723 is different enough from the 743/753 that it would be a bunch of work to support both. I have a handful of 144 pin H743ZI chips already.

rusefillc, 2022-01-07:
@mck1117 I am very slow. Are you available to make @NMSTEC an H7 proteus?

there are NUCLEO-H743ZI2 on aliexpress for relatively not crazy money. @NMSTEC your call on NUCLEO-H743ZI2 which would be same chip @mck1117 can solder on Proteus, i would maybe buy just one NUCLEO-H743ZI2 from aliexpress?

mck1117, 2022-01-07:
Having an H743ZI2 nucleo is a good idea in general.

I also have the chip to do another H7 proteus swap, and either @NMSTEC or I can do the soldering.