rusefi/rusefi #3763: Solenoid Clicking after data burn

BeerMoneyMotorsports, 2022-01-09:
While working on NB1, i have noticed the EGR boost solenoid starts clicking after doing a Burn.  This is very audible but does not see to cause performance issues.  It is output for INJ 6 per the schematics.  It controls ecu pins 4C and 3T.  

4D/3T are tied together as either can be used to control a boost controller.  Boost control isnt turned on in this tune, however the pin is selected in the menu.  

The solenoid will remain clicking until a full power cycle is done(ign and usb).  

Tune is https://rusefi.com/online/view.php?msq=813
Signature: rusEFI 2021.12.31.hellen-nb1.45104247

BeerMoneyMotorsports, 2022-01-09:
log https://rusefi.com/online/view.php?log=563

mck1117, 2022-01-10:
What kind of frequency?

Is this the pin set to boost control, but boost control is disabled?

BeerMoneyMotorsports, 2022-01-10:
i dont know exact frequency.  It clicks relatively fast.  

Correct, pin is set to boost control but it is not enabled.

mck1117, 2022-01-10:
not necessary, I know what's happening

mck1117, 2022-01-16:
@BeerMoneyMotorsports did this get fixed?

BeerMoneyMotorsports, 2022-01-16:
I have confirmed that i no longer have clicking with latest firmware

mck1117, 2022-01-16:
Great, will close then!

closed 2022-01-16 by mck1117