rusefi/rusefi #3772: console crash

rusefillc, 2022-01-11:
```
java.lang.IllegalStateException: While fan_control.cpp
	at com.rusefi.livedata.LiveDataParserPanel.refresh(LiveDataParserPanel.java:170)
	at com.rusefi.livedata.LiveDataParserPanel.<init>(LiveDataParserPanel.java:100)
	at com.rusefi.livedata.LiveDataParserPanel.createLiveDataParserPanel(LiveDataParserPanel.java:180)
	at com.rusefi.ui.LiveDataPane.<init>(LiveDataPane.java:44)
	at com.rusefi.ConsoleUI.<init>(ConsoleUI.java:138)
	at com.rusefi.StartupFrame.connectButtonAction(StartupFrame.java:278)
	at com.rusefi.StartupFrame.lambda$chooseSerialPort$0(StartupFrame.java:127)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.IllegalArgumentException: Negative length 137/136
	at com.rusefi.ui.livedata.Range.<init>(Range.java:18)
	at com.rusefi.ui.livedata.Range.<init>(Range.java:22)
	at com.rusefi.ui.livedata.Range.<init>(Range.java:27)
	at com.rusefi.CodeWalkthrough$1.colorStatement(CodeWalkthrough.java:153)
	at com.rusefi.CodeWalkthrough$1.enterExpressionStatement(CodeWalkthrough.java:62)
	at com.rusefi.livedata.generated.CPP14Parser$ExpressionStatementContext.enterRule(CPP14Parser.java:4195)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.enterRule(ParseTreeWalker.java:66)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:49)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at org.antlr.v4.runtime.tree.ParseTreeWalker.walk(ParseTreeWalker.java:52)
	at com.rusefi.CodeWalkthrough.applyVariables(CodeWalkthrough.java:46)
	at com.rusefi.livedata.LiveDataParserPanel.refresh(LiveDataParserPanel.java:156)
	... 43 more

```

rusefillc, 2022-01-11:
@mck1117 just for fun it's about
```
auto [cltValid, clt] = Sensor::get(SensorType::Clt);
```

mck1117, 2022-01-11:
https://en.cppreference.com/w/cpp/language/structured_binding

rusefillc, 2022-01-11:
https://github.com/antlr/grammars-v4/issues/2475 created

closed 2022-01-11 by rusefillc