rusefi/rusefi #3775: F7 ST bug work-around

rusefillc, 2022-01-11:
We can workaround #3566 on our end

mck1117, 2022-01-11:
I think adding an erase of sector 12 prior to the normal flash operation should work.

rusefillc, 2022-01-17:
``STM32_Programmer_CLI.exe -c port=usb1 -e 12``
sector 12 message does not look right but maybe it's just the message
[image]

``STM32_Programmer_CLI.exe -c port=usb1 -e 8``
sector 8 looks message looks nice but we know it's not getting the job done
[image]

rusefillc, 2022-01-17:
still verification error with both sectors 8 and 12 erased

[image]

rusefillc, 2022-01-20:
@mck1117 it look to me like I am unable to erase the sector we need to erase. See screenshots.

mck1117, 2022-01-24:
If I run `.\STM32_Programmer_CLI\bin\STM32_Programmer_CLI.exe -c port=usb1 -e [0 19]`, it says:

```
USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32  BOOTLOADER
SN          : 3057365D3438
FW version  : 0x011a
Board       : --
Device ID   : 0x0451
Device name : STM32F76x/STM32F77x
Flash size  : 2 MBytes (default)
Device type : MCU
Revision ID : --
Device CPU  : Cortex-M7

Erase sector(s) ...

erasing sector 0000 @: 0x08000000 done
erasing sector 0001 @: 0x08004000 done
erasing sector 0002 @: 0x08008000 done
erasing sector 0003 @: 0x0800c000 done
erasing sector 0004 @: 0x08010000 done
erasing sector 0005 @: 0x08020000 done
erasing sector 0006 @: 0x08040000 done
erasing sector 0007 @: 0x08060000 done
erasing sector 0008 @: 0x08080000 done
erasing sector 0009 @: 0x08084000 done
erasing sector 0010 @: 0x08088000 done
erasing sector 0011 @: 0x0808c000 done
erasing sector 0012 @: 0x08090000 done
erasing sector 0013 @: 0x080a0000 done
erasing sector 0014 @: 0x080c0000 done
erasing sector 0015 @: 0x080e0000 done
sector 0016 does not exist
sector 0017 does not exist
sector 0018 does not exist
sector 0019 does not exist
Flash page/sector erase command correctly executed.
Note: if flash sector is protected, it will not be erased.
```

But it has still not erased the content at 0x08080000!

(I filled the flash with 0xAA so I can see what's erased)

[image]

mck1117, 2022-04-11:
sure

closed 2022-04-11 by mck1117

rusefillc, 2022-08-05:
[image]

but 0x08080000 still not erased

[image]

rusefillc, 2023-06-30:
@mck1117 @nmschulte I wonder if a special custom version of https://github.com/rusefi/rusefi/blob/master/misc/install/STM32_Programmer_CLI/Data_Base/STM32_Prog_DB_0x451.xml would convince the tool of proper sector location

nmschulte, 2023-06-30:
I think that might work well actually, but I don't use these tools.