rusefi/rusefi #3779: Gear detection based on RPM to VSS ratio

rusefillc, 2022-01-12:
For stuff like #3007 do we need to figure out current gear based on RPM and VSS?

rusefillc, 2022-01-12:
@BeerMoneyMotorsports @blundar how are other ECUs detecting gear? any examples of relevant user facing settings?

BeerMoneyMotorsports, 2022-01-12:
I have 94-05 that “should “ have a vss input from cluster. I am in a position to confirm that signal in the next few days. I will log it and provide confirmation on my end.

mck1117, 2022-01-12:
I can wire a VSS from a front wheel on my Volvo if that helps (eventually)

NMSTEC, 2022-01-12:
Drive car at constant RPM in each gear, write down VSS.

RPM/VSS = Ratio ±5% error margin

For VSS calculation, XLS sheet attached.
[gearRatioCalc_DD.xls](https://github.com/rusefi/rusefi/files/7853418/gearRatioCalc_DD.xls)

BeerMoneyMotorsports, 2022-01-12:
screenshot of another ecu that uses TS for general information.

[image]

rusefillc, 2022-01-12:
@BeerMoneyMotorsports please help me see gear detection setting example, not boost control which uses detected gear

At the moment we are trying to have working gear gauge

BeerMoneyMotorsports, 2022-01-12:
here you go
[image]

rusefillc, 2022-01-14:
@BeerMoneyMotorsports what other options do they offer?

[image]

mck1117, 2022-01-14:
it may be for analog/digital sensors that detect the shifter position

BeerMoneyMotorsports, 2022-01-14:
[image]

rusefillc, 2022-01-14:
@BeerMoneyMotorsports @NMSTEC I can use help here so that I do not have to think. Need the formula to connect vehicle speed and gear ration

kilometers per hour * 1000  / 60 gives us meters per minute

meters per minute we divide by (2 * Pi * wheel diameter) to get wheel revolutions per minute

wheel revolutions per minute we divide or multiply by final drive ratio to get drivershaft revolutions per minute?

once we have  drivershaft revolutions per minute do we divide by engine PRM to get current transmission gear ratio?

NMSTEC, 2022-01-14:
Engine to WS: https://x-engineer.org/calculate-wheel-vehicle-speed-engine-speed/

In short
WS= (Circumference of tire *Eng RPM) / (Gear ratio * Final drive ratio)

mck1117, 2022-01-14:
driveshaft = wheel RPM * axle ratio (driveshaft spins faster than axle)
gearbox ratio = engine / driveshaft (engine spins faster than driveshaft)

rusefillc, 2022-01-15:
@BeerMoneyMotorsports  please test latest firmware

We are looking at "gearbox ratio"

you would need to configure two new settings at "Speed sensor config" dialog - finalGearRatio and wheelDiameter

NMSTEC, 2022-01-15:
Fun story, if harley going on dyno monday, I too can test it. This model has physical gearbox VSS, & I can try using RPMs as well.

BeerMoneyMotorsports, 2022-01-15:
First up is that final gear ratio will not accept the input ratio.  should be 4.3 but it reverts to 4.  

To note, final gear ratio is technically repetitive and was covered under speed sensor gear ratio.  

Is parameter for wheel diameter needed if we know revolutions per kilometer already?

FYI- I tested the old configuration yesterday and confirmed functionality on NB1.  

[image]

BeerMoneyMotorsports, 2022-01-15:
amending my last update, both final gear ratio and wheel diameter only want to accept whole numbers.

tune-  https://rusefi.com/online/view.php?msq=825
log- [vss.zip](https://github.com/rusefi/rusefi/files/7875990/vss.zip)

rusefillc, 2022-01-15:
@BeerMoneyMotorsports just made the following fixes
1)  final gear ratio and wheel diameter should now accept two digits after period
2) log entry name changed from s2rpm to Gauge Ratio

At the moment your vss.zip has something very weird in s2rpm maybe due to bad tune, let's try again once we have less crazy numbers un tune once we can enter those via TS

rusefillc, 2022-01-15:
https://github.com/rusefi/rusefi/issues/3792 created to track one potential setting duplicate

rusefillc, 2022-01-15:
#3793 created to track another potential setting duplicate

BeerMoneyMotorsports, 2022-01-16:
Ill attempt to test tomorrow but there is supposed to be snow

BeerMoneyMotorsports, 2022-01-17:
snow/ice delays

NMSTEC, 2022-01-17:
Harley going on Dyno today. Will be doing the VSS fuckery today. Forgot to figure out the VSS CAN byte.

rusefillc, 2022-01-26:
https://rusefi.com/online/view.php?log=603

[image]

rusefillc, 2022-01-26:
huh?

[image]

mck1117, 2022-01-26:
That's a downshift on the dyno, look at VSS at the same time, there's some jerky noise there. Potentially noise, potentially also actual jerk from the dyno.

Look how fuzzy VSS is in general:

[image]

BeerMoneyMotorsports, 2022-01-26:
quick run through a couple gears.  https://rusefi.com/online/view.php?log=615 
little longer drive with lots of gear changes.  https://rusefi.com/online/view.php?log=614

rusefillc, 2022-01-26:
just merged #3839 we would need to try it again once #3839 builds sorry my bad should have merged earlier

BeerMoneyMotorsports, 2022-01-30:
https://rusefi.com/online/view.php?log=626

rusefillc, 2022-01-31:
@mck1117 do we need different filtering parameters or else?

[image]

mck1117, 2022-01-31:
Huh? That log looks like clutch slip to me.

rusefillc, 2022-10-28:
@ElDominio can you please help with logs of gear detection which we apparantly have?

ElDominio, 2022-10-29:
Will test on tuesday using Lancer

rusefillc, 2022-11-13:
@ElDominio Tuesday?

ElDominio, 2022-11-14:
Tuesday from hell, the lancer is collecting dust, this should be easy but i am very lazy, I will eat a raw pinch of salt if I don't test this week

ElDominio, 2022-11-23:
> Tuesday from hell, the lancer is collecting dust, this should be easy but i am very lazy, I will eat a raw pinch of salt if I don't test this week

Today I tried to test on Raymond car, but I am trying to interpret VSS to feed to Gear logic. This is another comment to remind myself since it won't be finished today

ElDominio, 2022-12-07:
I got a log, it's gigantic.

[2022-12-06_19.56.42.mlg.zip](https://github.com/rusefi/rusefi/files/10171691/2022-12-06_19.56.42.mlg.zip)

The gear detection algorithm is confusing at least.

In basically everything I have used, all you do is make sure VSS reads correctly (calibrate VSS input to match speedo), then use VSS/RPM to determine gear (ECU saves how many revs are needed per mile for each gear, example: in first you can do 5mph/1000RPM, second 10mph/1000rpm... continues in picture)

[image]

mi-hol, 2023-02-14:
Just an example how other app handles this
[image]

rusefillc, 2023-03-15:
@ElDominio you are talking about improvements meaning you want to give me more work.

@ElDominio can we try a different approach? Would it be possible to get the current approach working first or is it totally unuseable?

ElDominio, 2023-03-15:
If we need to make this work, it needs to bw quite thoroughly explained. Understanding RPM/VSS is much easier

rusefillc, 2023-03-15:
@ElDominio please clarify

Do _you_ understand it enough to test or do _you_ need clarification?

@nmschulte @mck1117 @chuckwagoncomputing any volunteers?

mck1117, 2023-03-15:
@rusefillc you mentioned on fb some testing was needed.  I see perfectly functional gear detection both in unit tests and in real logs. What specific gap do you see?

rusefillc, 2023-03-15:
I have no resources to evaluate the data we have. If you declare gear detection working next step is documentation/case study/making it useable by @ElDominio ?

ElDominio, 2023-03-15:
[image]

Making it RPM vs VSS means we can use two pieces of data we already have, and assign a gear ratio to a gear.
I have prepared a spreadsheet explaining how it works. Please give it a look and feel free to comment
https://docs.google.com/spreadsheets/d/1x_v_taJ1OlW-1g6ioTMzR4uZp7twSlSOJcYXJrD6n4k/edit?usp=sharing

mck1117, 2023-03-15:
> I have prepared a spreadsheet explaining how it works.

Yeah, I know how it would work.

We do it that way because we need tire diameter/sensor params/gear ratio to make the regular VSS work. All of rusEFI tries to base the configuration on "real" numbers from first principles, not on magic fudge factors that you set with trial-by-error.  The real numbers are gear ratios and tire size.

rusefillc, 2023-03-15:
Both sides have a point

What if we set final drive ratio to 1 and wheel revolutions to 1000 would it allow the fudging approach via proper UI?

rusefillc, 2023-03-15:
@ElDominio does https://github.com/rusefi/rusefi/commit/6f7523fbef404e4eb404a71d0e6746a5acdafff9 help?

ElDominio, 2023-03-15:
> Yeah, I know how it would work.
>
> We do it that way because we need tire diameter/sensor params/gear ratio to make the regular VSS work. All of rusEFI tries to base the configuration on "real" [...]

My deal here is, knowing gear ratios depends on Google in basically all cases. Tire diameter is extremely inconsistent in an island where every single car has different wheel/tire combos and I can't have measuring tape to measure each individual car.  
This is the method I have used in both LinkECU and Haltech, which I know doesn't make it the correct one, but it seems to work well enough for pretty much every car I have come accross with.

The other question is, if I am required to use the current method, who explains how it works except telling me "get the numbers from the service manual)
___________________________________
Is setting ignition timing not trial-by error? Why can't VSS be like this?

mck1117, 2023-03-16:
@ElDominio you can set the final drive ratio to 1.0 or something, then look at this gauge (transmission section) to determine what the ratio should be in each gear

[image]

nmschulte, 2023-03-16:
I can't comment beyond that, naively, for my setup, I have determined the correct ratios and metrics for my purposes.  Here is a spreadsheet I use to compare my options, in case it helps anyone:
https://docs.google.com/spreadsheets/d/19XC9ctJKfD9I_oPKq4FVoLHUMUo3B_5aS0BMMmkeBjE/edit#gid=0

I think I can sympathize with @ElDominio's situation, though I've never been in it.  If nothing else, it might be helpful if rusEFI can aid to determine the ratio, such as @mck1117 suggests.  Personally, knowing the actual ratios is important to me, but perhaps not so much for others.

mck1117, 2023-03-16:
> Personally, knowing the actual ratios is important to me, but perhaps not so much for others.

it also helps with changing the mechanical parts out: if I change from a 4.10 to 3.73 rear end, I can change exactly one value that maps to that physical part of the car, and not have to re-tune my gear detection.

ElDominio, 2023-03-16:
Since my use case is mostly street cars, most people don't change out their ratios basically ever.

So, setting the final drive to 1 will spit out useable gear ratios for each gear?

I get wanting to do it with hard data but I get a gamut of cars in the shop, usually not dedicated race cars so changing the ratio is very uncommon. I get why you guys do it this way so if doing the final drive = 1 thing can get the gear detection working, that's a good enough fix for me

nmschulte, 2023-03-16:
I haven't messed with this part of the system yet, but for my application, there's a lot of assumption about the VSS (mainly, tire size and final drive ratio).  Anyway, assuming you have a trustworthy VSS speed, it seems like rusEFI could add command(s) to the firmware and buttons to Tuner Studio interface to compute the/a gear ratio for a given gear.  Possibly with some cleverness, it could automatically detect them all by shifting deterministically through all of the gears through a suitable range of RPM, when a certain mode is enabled.  This last idea would be more work than individual buttons, and both would be more work than what seems to already function (setting a "fake"/"guessed" final drive ratio and reading the computed gear ratio; see next para.).

@ElDominio, I gather you can use any value for the final drive to "spit out usable gear ratios," but using 1.0 seems sensible if you don't know the actual value.

mck1117, 2023-03-16:
> @ElDominio, I gather you can use any value for the final drive to "spit out usable gear ratios," but using 1.0 seems sensible if you don't know the actual value.

Yep, you can use whatever values you want for tire circumference and final drive.

This means your workflow stays the same.  For each gear, drive at some constant speed, and record the indicated `Gearbox Ratio` in the field for that gear.

rusefillc, 2023-09-17:
looks to be working

closed 2024-02-17 by rusefillc