rusefi/rusefi #3780: 't' in terminal does not print seconds properly

rusefillc, 2022-01-12:
connect to rusEFI unit via terminal
hit 't' button a few times
expected behavior: testCommandCounter increases, seconds show number of seconds since ECU start

actual behavior:
[image]

rusefillc, 2022-01-12:
At first I've suspected that ``getTimeNowSeconds`` is not returning expected int value but TS seems not to be affected

[image]

is that some %d issue with 64 value on stack not 32 bit value? sometime else?

rusefillc, 2022-01-12:
@racer-coder any chance you could look into this random bugreport?

mck1117, 2022-01-12:
%d can't print a 64-bit

commit 0f38a22a (2022-01-12): 't' in terminal does not print seconds properly fix #3780
closed 2022-01-12 by rusefillc