rusefi/rusefi #3782: Console Freezing, disconnecting, then crashing

NMSTEC, 2022-01-12:
Situation & problem:

Download snapshot F4 proteus bundle. 
Upon connecting to F4 with console, console loads, works properly for 3 seconds, proceeds to change up top to "COM 6 Not loaded", then freeze for around 20 seconds, then refreshes and shows "not connected". Another 10 seconds later, just closes.

rusefillc, 2022-01-12:
I've created a random snapshot https://github.com/rusefi/rusefi/releases/tag/untagged-ffa6366882a79b8a0336 just to make sure we use same files

rusefillc, 2022-01-12:
17 seconds from ECU start
[image]

rusefillc, 2022-01-12:
37 seconds from ECU start everything works

[image]

unable to reproduce issue

NMSTEC, 2022-01-12:
Things attempted:

Replaced cable. -Same problem
Connected via putty. -Works fine
Using old console. -Works fine

Log attached
[MAIN_rfi_report_2022-01-12_12_42_10_862.txt](https://github.com/rusefi/rusefi/files/7857696/MAIN_rfi_report_2022-01-12_12_42_10_862.txt)

NMSTEC, 2022-01-12:
This seems to only affect one of my PCs. 100% console update related & windows, but unknown reason. 
Second PC has 0 issues with the same exact setup.

NMSTEC, 2022-01-14:
Laptop is now having same issue as First PC. Using snapshot FW, connected to Harley, Proteus F7. Tunerstudio nicely connects and talks. Connect console, loads up, and freezes a few seconds in.

mck1117, 2022-01-14:
can you `java -v`?

NMSTEC, 2022-01-14:
https://github.com/rusefi/rusefi/releases/tag/2021.12.22_release
This release works perfectly fine

[image]

NMSTEC, 2022-01-15:
This is driving me crazy. Problem proceeds. Just grabbed another laptop with W10. Same problem. Fully virgin computer.

PC #1 W11 - Works perfectly
PC #2 W10 - Only works with December console.
PC #3 W7 - This is the laptop that i've been using for rusEFi the entire time. Only works with December console.
PC #4 W7 - Only works with December console.

The cable used for all of these is the same. Have also attempted replacing with known good SHORT cable. Same problem.

Serial console (Putty) has no issues communicating with the ECU on any PC.

Tuner studio has no issues communicating with ECU on any PC.

All Java versions are the same.

I'll look for which revision specifically broke this intermittent console problem.

rusefillc, 2022-01-15:
Let me test with w7

So far only tested w11

rusefillc, 2022-01-16:
W10 works good for me

shadowm60, 2022-01-22:
w10 not working on my side as well
https://i.imgur.com/YYgzocy.png

December release is working and connects to the target

mck1117, 2022-01-22:
@NMSTEC exactly which version DID work for you when you went back in time?

NMSTEC, 2022-01-22:
December Release.

NMSTEC, 2022-01-23:
#3826 #3827

shadowm60, 2022-01-31:
Any updates? i was testing nightly on mre and console crashed, https://i.imgur.com/dkglnlh.png

rusefillc, 2022-01-31:
Root cause: in rare combinations of mismatched console and firmware versions console expects more outputs than firmware is willing to return and that does not go well

proper fix would be #3827

in the meantime users should be more careful mixing and matching incompatible firmware and console.

rusefillc, 2022-01-31:
@shadowm60 i guess your issue is probably different

please attempt release https://github.com/rusefi/rusefi/releases/tag/20220126.release and please double check that firmware and console come from the same binary

if you are having an issue please create a separate ticket

closed 2022-01-31 by rusefillc