rusefi/rusefi #3784: no start after firmware snapshot update

BeerMoneyMotorsports, 2022-01-12:
Car- NB1
ECU- Hellen NB1 rev B
Problem- After updating firmware via console, the car no longer starts.  I can revert back to previous firmware and car will start
New Firmware- rusEFI 2022.01.12.hellen-nb1.2528379042
Previous Firmware- rusEFI 2021.12.31.hellen-nb1.45104247
Tune- https://rusefi.com/online/view.php?msq=820
Log- https://rusefi.com/online/view.php?log=565
Screenshot of warning in console.  TS shows "NO IGNITION" and "NO INJECTION"  while cranking.
[image]

rusefillc, 2022-01-12:
I had this issue on passat. I've implemented limp code gauge. It looks like configuration memory mapping is broken somewhere.

mck1117, 2022-01-12:
yeah, and that limp code does indeed say 3, which is rpm limit (as the warning suggests)

mck1117, 2022-01-13:
@rusefillc I'll say I disagree with trying to compress scalars (as opposed to tables) like this:
`uint8_t autoscale rpmHardLimit;...;@@RPM_1_BYTE_PACKING_MULT@@, 0, 0, 20000, 2`

rusefillc, 2022-01-13:
yep, 6k limit or even 12k limit is too low for RPM.

rusefillc, 2022-01-13:
while I agree that 1 one RPM is a mistake what is the exact issue in this exact spot right now? looks unsigned is not it supposed to survive 256*50?

mck1117, 2022-01-13:
well, it should only be storing 140 since beer's rev limit is 7000 (7000/50 -> 140), but for some reason the limp manager is seeing 0

rusefillc, 2022-01-13:
140 > 128, signed/unsigned drama somewhere maybe? just guessing

mck1117, 2022-01-13:
the flash format hasn't changed since the known-good firmware, it already had the change to rpm limit format in place

BeerMoneyMotorsports, 2022-01-13:
[no_start_nb1.zip](https://github.com/rusefi/rusefi/files/7858888/no_start_nb1.zip)

mck1117, 2022-01-13:
TS trying to write to the rev limit
[image]

notice that the debugger thinks the rev limit is at `persistentState+10` and TS is writing to `persistentState+13`

mck1117, 2022-01-13:
found it https://github.com/rusefi/rusefi/commit/364783f65851c3b1fa3d0f36eb1a0eac306202cc#diff-99378c9c020bb94c16fc2dc8fc66e709f06a0ad873b0a6ad23e94c88f29d5fb4