rusefi/rusefi #3786: assert that C and TS enum sizes match

rusefillc, 2022-01-13:
At the moment we rely on humans asserting that enum size matches between C and TS

we shall add assertions to confirm that constraint

mck1117, 2022-01-13:
We care about enum size, but we care more about offset being correct. The problem here was that the different enum size caused later elements to move around in C, but not in TS.

We can pretty easily generate something that looks like
```
static_assert(offsetof(engine_configuration_s, engineType) == 0);
static_assert(offsetof(engine_configuration_s, sensorSnifferRpmThreshold) == 4);
static_assert(offsetof(engine_configuration_s, rpmHardLimit) == 5);
```

rusefillc, 2022-01-13:
great point, yes we can generate assertions for all offsets that's a great idea!

at the moment I am failing to get
```
#define engine_type_e_size 4
custom engine_type_e @@engine_type_e_size@@ bits, S32, @OFFSET@, [0:6], @@engine_type_e_auto_enum@@
```
to get antlr partsed

rusefillc, 2022-04-30:
We have assertions now

closed 2022-04-30 by rusefillc