rusefi/rusefi #3792: Replace "wheel revolution per km" with wheelDiameter

rusefillc, 2022-01-15:
in my opinion humans should not be entering "wheel revolutions per km" in my opinion humans should be entering something more natural

hopefully we can agree that wheelDiameter is better than "wheel revolution per km"?

BeerMoneyMotorsports, 2022-01-15:
Im fine with either. Both can be located with a quick google search and thats what users will do.

mck1117, 2022-01-15:
Most factory computers and all tire size calculators use or output rev per km

rusefillc, 2022-01-15:
@mck1117 so most factory computers and all tire size calculators use or output rev per km
and #3793 says that we prefer natural units.

is there a contradiction here? does #3793 argument tells us to throw away output rev per km?

dron0gus, 2022-01-16:
Most user-friendly will be input like "205/55R16" with magic math happening inside TS or ECU.
While we still need "pulses per wheel revolution" for gear detection.

mck1117, 2022-01-16:
the trouble is that the "pure" tire size math is usually slightly wrong since tires are bendy

dron0gus, 2022-01-17:
> the trouble is that the "pure" tire size math is usually slightly wrong since tires are bendy
Yes. Pure tire size for accurate speedo and some algos that needs real car speed (like fan disable on high speed).
While pulses per revolution for gear detection.