rusefi/rusefi #3793: replace speed sensor gear ratio

rusefillc, 2022-01-15:
there are many different implementations of vehicle speed sensor - some are CAN, some are tooth on drivestaft side, some are tooth on wheel.

sometimes we need more natural final drive ratio

overall open question which settings are most suited when and what kind of duplication do we have if any

[image]

mck1117, 2022-01-15:
Speed sensor gear ratio may not always be axle ratio: for example, using ABS sensor for speed sense or diff-mounted speed sensor would have speed sensor ratio of 1.0. My setup will have 48t, 1.0 ratio for speed sense, but 3.73 from axle to gearbox output

rusefillc, 2022-01-15:
Should we replace these two
[image]

with one "sensos tooth to wheel revolution ratio" and use it along wide wheel diameter?

mck1117, 2022-01-15:
I much prefer quantities that users can actually measure or inspect without math. I can count my sensor teeth and gear ratio, and type those in and there is zero opportunity for wrong math. It's a great opportunity for us to differentiate ourselves with an easier user experience.

rusefillc, 2022-01-15:
I like "actually measure or inspect without math"

commit d2245c1c (2022-01-15): replace speed sensor gear ratio fix #3793
closed 2022-01-15 by rusefillc