rusefi/rusefi #3794: Migrate to Chibios 21 or newer

rusefillc, 2022-01-16:
As of today rusEFI uses ChibiOS 20. One day we shall use newer ChibiOS, today version 21 is available

rusefillc, 2022-08-18:
@Finist64 please fork both https://github.com/rusefi/rusefi and https://github.com/rusefi/ChibiOS

See also https://github.com/rusefi/rusefi/pull/4286
See also https://github.com/rusefi/rusefi/pull/4303

rusefillc, 2023-02-13:
Some nice commits at https://github.com/rusefi/rusefi/tree/chibi-21

Some nice commits at https://github.com/mck1117/rusefi/tree/chibi-21

rusefillc, 2023-08-16:
Status update: F4 compiles but does not start for unknown reason

best commits at https://github.com/rusefi/rusefi/pull/5393
less nicely done commits at https://github.com/rusefi/rusefi/pull/5394

dron0gus, 2025-07-04:
Trying one more time https://github.com/rusefi/rusefi/pull/8197

closed 2025-07-06 by rusefillc