rusefi/rusefi #3803: Better handling of RPM during stop of cranking

rusefillc, 2022-01-18:
At the moment failed attempts to start engine produce really confusing noisy log files

There must be a few low hanging fruits to make logs of stop of cranking less confusing

B6 Passat for instance. Note 2 second of 430 RPM logged at the end of cranking attempt

[image]

rusefillc, 2022-01-18:
Relevant magic constants:
NO_RPM_EVENTS_TIMEOUT_SECS

mck1117, 2022-01-18:
Worst case engine has 1 tooth per crank rev (let's assume we don't care about 1 tooth per cam rev engines, if those exist?).

That means at 100 rpm, that tooth arrives every 0.6 second.  So we can probably reduce `NO_RPM_EVENTS_TIMEOUT_SECS` from 2 to 1, at least.

racer-coder, 2022-01-18:
Make it trigger dependent?  People with 60-2 shouldn't suffer a poor experience just because others have a crappy trigger setup.

rusefillc, 2022-01-18:
great minds thinks alike

[image]

mck1117, 2022-01-18:
> Make it trigger dependent? People with 60-2 shouldn't suffer a poor experience just because others have a crappy trigger setup.

You don't necessarily want it _too_ low, lest it rapidly cycle something on/off at the start/end of cranking

rusefillc, 2022-01-20:
progress was made!

closed 2022-01-20 by rusefillc