rusefi/rusefi #3804: On m_event vs lobe vs scheduling

rusefillc, 2022-01-18:
We have one m_event and three lobes

We either have a bug with events being constantly rescheduled and rarely actually happening due to m_event reuse or we have some smart logic which really makes sure that previous event is done before next one is scheduled/cancelled.

Assuming we have that smart logic can we please add helpful code comments clarifying the magic?

racer-coder, 2022-01-18:
The lobes are non-overlapping.  We can use 1 instead of 3 for the same reason we don't need separate m_event for turning the solenoid on and off.  The callback that twiddles the pin also schedules the next event.  It should be really obvious from the code:

void HpfpController::pinTurnOn(HpfpController *self) {                                              
        enginePins.hpfpValve.setHigh();                                                             
                                                                                                    
        // By scheduling the close after we already open, we don't have to worry if the engine      
        // stops, the valve will be turned off in a certain amount of time regardless.              
        scheduleByAngle(&self->m_event.scheduling,                                                  
                        self->m_event.scheduling.momentX,                                           
                        self->m_deadtime + engineConfiguration->hpfpActivationAngle,                
                        { pinTurnOff, self });                                                      
}                                                                                                   
                                                                                                    
void HpfpController::pinTurnOff(HpfpController *self) {                                             
        enginePins.hpfpValve.setLow();                                                              
                                                                                                    
        self->scheduleNextCycle();                                                                  
}

mck1117, 2022-01-18:
> The callback that twiddles the pin also schedules the next event

seems obvious enough to me

closed 2022-01-18 by mck1117

rusefillc, 2022-01-18:
``The callback that twiddles the pin also schedules the next event. `` this was NOT obvious enough to me, this is the part where better comments were desired. I've pushed somethng.

racer-coder, 2022-01-18:
One thing I will say is that this only works if there is one tooth between the time we call scheduleOrQueue and the time the event is supposed to occur.  This means, roughly, for a 3 lobe system you need 6 or maybe 12 teeth trigger.  On a 60-2 this is of course no problem.

I don't know if all GDI engines have a similar high-tooth count trigger.  I think 60-2 has been the spec for BMW for 25+ years, maybe it's a German/Bosch thing (which maybe is why you see it on VW too).  If you want this to work on low tooth triggers without reducing the responsiveness (i.e. without having to wait a full rotation to change how long the pump runs) you'd have to plumb through the last known tooth somewhere.  IMO the trigger system itself should know the timing of the last tooth so the caller doesn't have to plumb it through as an extra argument to scheduleOrQueue, but that of course is a bigger change...  If you do that you may want to plumb angle in absolute terms (since the start of the ECU) instead of relative terms (to the last engine cycle) to avoid confusion over whether scheduling something in 1 degree means "really soon" or one engine cycle away, and ... blah blah blah hence, a bigger change.....

racer-coder, 2022-01-18:
(and no I'm not volunteering)

mck1117, 2022-01-18:
All modern engines I can think of use 60-2. My mazda does, all bmw/mercedes/vw/other bosch has for 20+ years, GM uses it since 2005-ish, ford since a similar time.

rusefillc, 2022-01-19:
@mck1117 @racer-coder assertion added https://github.com/rusefi/rusefi/commit/f470d93cca3cc257fcdddaf8bcfaac5b9a0b2975