rusefi/rusefi #3805: gcov coverage seems stale

rusefillc, 2022-01-18:
https://rusefi.com/docs/ points at https://rusefi.com/docs/unit_tests_coverage/ which says

Date: | 2021-12-04 07:39:23

rusefillc, 2022-01-18:
https://github.com/rusefi/rusefi/blob/master/.github/workflows/build-unit-tests.yaml is the job

rusefillc, 2022-01-18:
[image]

rusefillc, 2022-01-18:
@chuckwagoncomputing I've made at least one fix but docs/unit_tests_coverage does not seem better :( can you please help?

[image]

chuckwagoncomputing, 2022-01-20:
Doesn't seem like this is fixed yet...

rusefillc, 2022-01-20:
@chuckwagoncomputing date looks better for me

[image]

rusefillc, 2022-01-20:
I vote to close? @chuckwagoncomputing maybe caching issue on your end?

closed 2022-01-20 by rusefillc

chuckwagoncomputing, 2022-01-20:
That date/time is just now. Not sure what happened exactly. I _thought_ I reloaded the page on my end.