rusefi/rusefi #3806: sporadic HighRevTest testV12

rusefillc, 2022-01-18:
GHA has one failure and then recovered

Jenkins had about six failures so far and recovered

[image]

rusefillc, 2022-01-21:
before big RPM change

[image]

rusefillc, 2022-01-21:
@mck1117 any chance you could have a look?

rusefillc, 2022-01-23:
Acceptance criteria: 30 consecutive successful builds on Jenkins (at 11 minutes per job on Jenkins that's 7 hours of testing unfortunatelly)

I've disabled master job and will start binary search. Will keep this ticket updated.

rusefillc, 2022-01-24:
looks like https://github.com/rusefi/rusefi/commit/9f5188bf305a2327f957548759e9606f9b3cccfa is the cause

plan to cache this calculation

rusefillc, 2022-01-25:
@mck1117 still an issue :( https://github.com/rusefi/rusefi/actions/runs/1743220941

rusefillc, 2022-01-25:
[image]

rusefillc, 2022-01-25:
that commit which returns math back to 1 second has helped jenkins

[image]

closed 2022-01-26 by rusefillc

mck1117, 2022-01-26:
#3838 says we are probably OK