rusefi/rusefi #3812: Bosch Quick Start on B6 Passat is confused about phase

rusefillc, 2022-01-19:
See log attached to https://rusefi.com/forum/viewtopic.php?p=44114#p44114

exhibit one - revolution counter increase at 2.235 vvtSyncGapRatio
[image]

exhibit two - revolution counter increases at 1.556 vvtSyncGapRatio
[image]

rusefillc, 2022-01-19:
Other triggers tell us to use redundant sync pattern recognition with extra gaps which we do not do here - as is we use single gap length of 1 which means that lucky noise would be treated as sync point

current state:
```
	int offset = 20;
	s->setTriggerSynchronizationGap3(0, 2, 3);
	s->addEvent360(offset + 0, T_PRIMARY, TV_RISE);
```

rusefillc, 2022-01-19:
OMG :( we were totally not using VVT info of VVT_BOSCH_QUICK_START and VVT_NISSAN_MR and etc etc

```
	switch (engineConfiguration->vvtMode[camIndex]) {
	case VVT_FIRST_HALF:
	case VVT_MAP_V_TWIN_ANOTHER:
		return syncAndReport(tc, getCrankDivider(operationMode), 1);
	case VVT_SECOND_HALF:
		return syncAndReport(tc, getCrankDivider(operationMode), 0);
	case VVT_MIATA_NB2:
		/**
		 * NB2 is a symmetrical crank, there are four phases total
		 */
		return syncAndReport(tc, getCrankDivider(operationMode), 0);
	case VVT_NISSAN_VQ:
		return syncAndReport(tc, getCrankDivider(operationMode), 0);
	default:
	case VVT_INACTIVE:
		// do nothing
		return 0;
	}
```

@racer-coder FYI

mck1117, 2022-01-19:
@rusefillc can you capture a logic analyzer trace of both the crank and both cam signals for a few cranking attempts (same log is fine, we can cut it up)?  Would be a great test case to have like we already have for Miatas and LS.

rusefillc, 2022-01-19:
@mck1117 yes huge change of logic analyzer trace in the next 48 hours

mck1117, 2022-01-19:
> @mck1117 here you are logic analyzer data [rusefi/rusefi_documentation@f9c935a](https://github.com/rusefi/rusefi_documentation/commit/f9c935afaad0c275b3ddd98e20324c0a10108fa4)

these traces are very very noisy on the vvt channel, what sort of grounding did you use for the logic analyzer?

rusefillc, 2022-01-19:
Hook to pin soldered to bolt ring

[image]

racer-coder, 2022-01-19:
TunerStudio logs fast enough that you can capture the cam and hpfp pins.  300 RPM = 1.8 crank degrees/ms.  The 3-5ms period would be 5.4-9 degrees per log line, enough to get a rough idea where the cam is relative to hpfp firing.  If you had a counter on crank trigger pulses then you could get a rough idea where that is too.

Probably not good enough for writing unit tests, but enough to debug field trigger issues.

rusefillc, 2022-01-20:
5 degrees of VVT position swing on cranking seems like progress to me :) that's with second gap ratio in pattern definition

[image]

[2022-01-19_21.22.14.zip](https://github.com/rusefi/rusefi/files/7901801/2022-01-19_21.22.14.zip)

closed 2022-01-20 by rusefillc