rusefi/rusefi #3828: GM 2 BAR MAP sensor values wrong

duhafnusa4, 2022-01-23:
the kpa and voltages for the default GM 2 BAR MAP sensor are actually the values for a GM 3 BAR sensor. 

Below are the Values for the 1,2 and 3 BAR sensors (we should make all 3 defaults) 

[image]

duhafnusa4, 2022-01-23:
Theres already one on the default list called "3 bar". It should just be re-named "GM 3 Bar", and then correct the values for the 1 and 2 BAR GM

rusefillc, 2022-01-23:
@duhafnusa4 thank you! Just renamed "3 bar" to "GM 3 Bar"

rusefillc, 2022-01-23:
kind of reminds me of #3054

rusefillc, 2022-01-23:
@duhafnusa4 why do you think that 2 bar settings are wrong?

how do you know that? what are you doing where and how to see rusEFI not working properly?

reason I am asking for clarification is due to fact that code looks OK. I wonder if calibration is right but we have some sort of usability confusion

```
	case MT_GM_3_BAR:
		return &gm3bar;
	case MT_GM_2_BAR:
		return &gm2bar;
	case MT_GM_1_BAR:
		return &gm1bar;

static FastInterpolation gm3bar(0.631, 40, 4.914, 304);

static FastInterpolation gm2bar(0, 8.8, 5, 208);

static FastInterpolation gm1bar(0, 10, 5, 105);

```

duhafnusa4, 2022-01-23:
Yeah kind of inconsistent. We dont need 4 billion different presets. People can always do custom if they want. I would say if you have a hellen board, then we should have defaults for that chassis' sensors

rusefillc, 2022-01-23:
@duhafnusa4 you are starting to mix more and more sexy stuff into one overloaded issue

let's go back to original problem statement. what makes you report this issue? please elaborate

duhafnusa4, 2022-01-23:
> @duhafnusa4 why do you think that 2 bar settings are wrong?
>
> how do you know that? what are you doing where and how to see rusEFI not working properly?
>
> reason I am asking for clarification is due [...]

see the values for 1 bar:

[image]

see the values for 2 bar:

[image]

rusefillc, 2022-01-23:
greyed out values do not mean shit. that's a usability issue - you are looking at greyed out values which are only use for 'custom' and you assume that greyed out values are applied, those are not

this would be addressed if we make progress with #3054 - that's a usability issue not functional issue.

duhafnusa4, 2022-01-23:
even if greyed out we should be displaying the values for that sensor.

rusefillc, 2022-01-23:
@duhafnusa4 I agree that current state of affairs is shit

rusefillc, 2022-01-23:
@BeerMoneyMotorsports @OrchardPerformance @NMSTEC significant part of this change is rinse and repeat https://github.com/rusefi/rusefi/pull/3326/files