rusefi/rusefi #3837: interactive pinout - can we merge sections?

rusefillc, 2022-01-25:
@chuckwagoncomputing would it be OK for me to merge these two sections with three classes for pin: 36?

that's microRusEFI pin reuse, we have similar duplication for four pins

```
  - pin: 36
    id: [GPIOB_0, EFI_ADC_8]
    class: [switch_inputs, analog_inputs]
    color: orange left    
    ts_name: 36 - AN volt 8, Aux Reuse
    type: av

  - id: GPIOB_0
    class: outputs
    ts_name: AUX AV8 reuse

```

chuckwagoncomputing, 2022-01-25:
As far as the pinouts are concerned, yes that's fine. Won't it break the configs of people who are using AUX AV8 reuse though?

rusefillc, 2022-01-26:
@chuckwagoncomputing true

it's a trade-off for sure but it is important to emphasize same wire under same name instead of same wire under different names

commit cccc51a6 (2022-01-26): microRusEFI - pinout - can we merge sections fix #3837
closed 2022-01-26 by rusefillc