rusefi/rusefi #3841: well how many ECUs to driver N73

rusefillc, 2022-01-26:
What do we do about two HPFPs? two ECUs or support two units?

rusefillc, 2022-01-28:
My preference is really to run one ECU on the complete N73 engine. I believe at least audi V10 is two pumps as well?

@racer-coder any chance you would be interested to make rusEFI support two pumps on one ECU?

racer-coder, 2022-01-28:
Not interested; the hard work isn't figuring out how to duplicate the HPFP work (just instantiate the class twice, similar thing is already done for Etb; you may want/need to use std::array<> due the use of EngineModule).

The hard work is deciding how RusEFI handles multiple banks.  Separate MAP and thus fuel load per bank?  Plumb that into each HpfpController separately.  Duplicate pin selection for HPFP and pressure sensor, make it optional depending on how many banks etc.  A bunch of UX decisions....

mck1117, 2022-01-28:
> The hard work is deciding how RusEFI handles multiple banks. Separate MAP and thus fuel load per bank? Plumb that into each HpfpController separately.

As of recently we actually calculate per-cylinder fueling (as opposed to per-engine or per-bank), so the change from 1 to N fully independent fuel calculations actually wouldn't be that hard.

mck1117, 2022-01-28:
We could pretty easily add a second HPFP setup that uses the same pressure target as the first, as that scenario does also exist (single plenum, dual fuel systems). That would run a V12 well enough as a test, and maybe combined with bank-to-bank balancing (automatic or manual) would be just as good as doing full independent per-bank calculations.

racer-coder, 2022-01-28:
Yeah I would think separate PID would be enough to paper over air flow differences between banks.

mck1117, 2022-01-28:
Which separate PID? You certainly need independent pump controllers just to deal with stuff like manufacturing tolerance in each pump/injectors/plumbing/etc. The only thing "wrong" on the slave bank (ie, the one whose MAP sensor we're not using) would be the fuel demand, and will at least be close enough to correct.

racer-coder, 2022-01-28:
Separate PID as in separate controllers (separate sensors, separate accumulated I, etc).  Not separate TunerStudio config for P and I.
We're in agreement here.

racer-coder, 2022-01-28:
BTW if you want to muddy the waters, the BMW S55 engine (2014-2019 M3/M4 and some later M2) has two HPFP but one rail.  One pump is used for low load, and the second pump is activated at higher loads.  It (like the BMW N55 - 2012-2019 335i for example) has a separate shaft to run the pumps, which has its own "camshaft" position sensor.

mck1117, 2022-01-28:
I think it's a long time before we run an S55 :)

billstewartx, 2023-03-26:
> The hard work is deciding how RusEFI handles multiple banks. Separate MAP and thus fuel load per bank?  

Note: VVL valvetronic does not have a MAP sensor... it is a MAF, since the manifold pressure is at Atmospheric most of the time