rusefi/rusefi #3854: Console: "Pause" button not working?

andreika-git, 2022-01-29:
It seems that the "Pause/Resume" button stopped working.
Expected: When the "Pause" is pressed, the console messages stop updating.
Actual: new messages are added...
[image]

P.S. The console is compiled from the current master.

mck1117, 2022-01-30:
works for me?

rusefillc, 2022-02-01:
@mck1117 does not work for me - note that there are two "pause" buttons on the same screen, it's the left-top which is does not work

closed 2022-02-01 by abelom

rusefillc, 2022-02-01:
it's not a great UI :( top-left button pauses gauges while the other button pauses text messages

I've renamed the top-left button to reduce the confusion a bit. thank you @andreika-git

commit ce334f07 (2022-02-01): Console: Main "Pause" button not working? fix #3854