rusefi/rusefi #3857: Feature request: odometer for engine break

rusefillc, 2022-01-30:
@NMSTEC is asking to have custom calibration for engine break-in period

that would depend on at least a rough somewhat precise odometer which raises the question of persisting rough odometer value

rusefillc, 2022-01-30:
Here's a proposal: for ECUs which control their own power cycle we can use stm32 internal flash. in order to reduce risks we can designate a full page of internal flash just for odometer

in order to reduce waste of flash we can go with the small 16K pages. Even if we use two of those for bootloader we still have two more of these small pages to use for odometer

NMSTEC, 2022-01-30:
I'd need that eeprom/internal flash page to store a number of different things in it, including VIN number (Part of security), User info, and etc. 

I'd still push an eeprom if it does not interfere with anything, as it is meant exactly for this, and is decently easy to implement.

rusefillc, 2022-01-30:
@NMSTEC we need exact requirements before we decide to add new type of hardware. At the moment too much is only in your head and not in issues. The stuff I am aware of seems to work great with 16K of stm32 flash

NMSTEC, 2022-01-30:
16k is absolutely plenty for my stuff. My only concern is a full erase by accident/on purpose wiping out all the information. Unless we make a custom console to prevent that from happening.

mck1117, 2022-01-31:
@rusefillc this needs external eeprom/fram/spi flash so that you can't easily erase it (especially by accident)

mck1117, 2022-01-31:
Also, all the small pages are at the bottom, and it's a huge PITA to try and build a firmware image with a hole in it.

dron0gus, 2022-02-01:
Note, that there is not need to erase whole page each time. We can "append" updated data to unwritten area until page end.
For example, if we allocate 512 bytes for odometer and other stuff we can write in 16k/0.5 = 32 times untill we need to erase page. 
Also using two pages we can be safe to erase second page while keeping backup (previous data) in first page.
But, please, lets use ChibiOS Flash driver and Managed Flash Storage instead of writing anything new.
I used it for SPI flash on my custom RE board and it looks good for me.