rusefi/rusefi #3868: MSD vs SD card logging is confusing as hell

rusefillc, 2022-02-01:
I am tired of wasting my time reading same user hand holding again and again

There are clearly two signals here:
1) this has to be documented unless see below
2) we need this to _somehow_ become way less confusing

rusefillc, 2022-02-01:
[image]

rusefillc, 2022-02-01:
worst case scenario we would have to kill this smart logic and implement some way more understandable logic with a button or else

I love the current logic but user support noise is a clear signal that change is needed

opelpanfan, 2022-02-01:
Not sure if that is a good approach, but sd card logging is stopped if no SPI is selected. That is the case with the SDIO interface.
[image]

rusefillc, 2022-02-01:
@opelpanfan that's a great separate issue! please create a separate issue for the very valuable think you've noticed!!!