rusefi/rusefi #3887: rusefi_console connection issue

shadowm60, 2022-02-03:
Running on W10

java version: 
java -version
java version "1.8.0_321"
Java(TM) SE Runtime Environment (build 1.8.0_321-b07)
Java HotSpot(TM) Client VM (build 25.321-b07, mixed mode)

starting console command line with:
java -jar rusefi_console.jar > log_java_updated
I 220203 194741.694 [main] Launcher - rusEFI UI console 20220124
I 220203 194741.717 [main] Launcher - Compiled Tue Feb 01 21:08:46 EET 2022
I 220203 194741.717 [main] Launcher -

ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
line 8:6 missing ';' at '['
[log_java_updated.zip](https://github.com/rusefi/rusefi/files/7997012/log_java_updated.zip)

Connection and configuration with tuner studio is working fine.
Next i will try also with linux.

Default tune loaded after firmware update to avoid faulty tune from my old project.
Used board was MRE 0.3 F7
snapshot_20220126_224458_mre_f7_rusefi 
Updated via rusefi_console, log attached

[20220126_224458_update.zip](https://github.com/rusefi/rusefi/files/7997035/20220126_224458_update.zip)

rusefillc, 2022-02-03:
@shadowm60 I see a lot of details but I do not see problem statement? what's wrong? when?

shadowm60, 2022-02-03:
Connection is not made, console closes after a timeout.

rusefillc, 2022-02-03:
@shadowm60 are you sure port is not used by TS?

are you pretty much saying that console just unconditionally totally not working for you?

shadowm60, 2022-02-03:
Yes. I did a power on of the board, testws console with the attached output i catched. Then i opened ts, and comnection was working from that tool.

Befor this i did a java update since it was crashing and i was wondering if old java is the issue or console.

None of my tricks worked.

Should i get a nightly and try that as well? Or nothing was changed since january release.

shadowm60, 2022-02-03:
with an older version of java i had a bit different log:
java -version
java version "1.8.0_281"
Java(TM) SE Runtime Environment (build 1.8.0_281-b09)
Java HotSpot(TM) 64-Bit Server VM (build 25.281-b09, mixed mode)
[log.zip](https://github.com/rusefi/rusefi/files/7997211/log.zip)

rusefillc, 2022-02-03:
do you have another PC to try on?

java crashing sounds very unusual

shadowm60, 2022-02-03:
[log_linux.zip](https://github.com/rusefi/rusefi/files/7997359/log_linux.zip)
log from my linux laptop, i noticed this but probably not the root cause:
2022-02-03_20_44_52_944<EOT>: Settings file not found or damaged. Using default settings.
2022-02-03_20_44_59_370<EOT>: Console 20220124
2022-02-03_20_44_59_383<EOT>: Hardware: F4

i have F7 on mre.

ttyACM0 -> tested with TS, and it was working, so port is available for the user and usable

shadowm60, 2022-02-03:
Tested my w10 with Frankenso and snapshot_20220126_213825_frankenso_na6_rusefi and got the same output/disconnection.

Can i gather more data? or how can i help?

rusefillc, 2022-02-03:
please download and install https://www.jetbrains.com/idea/download/ community edition

we would need to do remove screen sharing to debug not today

mck1117, 2022-02-16:
This is likely fixed by #3936. @shadowm60, can you test with latest snapshot console?

shadowm60, 2022-02-16:
I can, can you say if the snapshot for f7 mre contains this? Noticed that not all the snapshots are with the same date, i can test ok f4 disco as well, but i need it to work on mre f7. Will come back with feedback today

shadowm60, 2022-02-16:
Tested console with snapshot_20220216_063719_mre_f7_rusefi and works fine.

2022-02-16_11_25_53_279: EngineState: USB: got Query command
2022-02-16_11_25_53_279: EngineState: got S/H (queryCommand) mode=1
2022-02-16_11_25_53_279: EngineState: Primary UART RX
2022-02-16_11_25_53_280: EngineState: Primary UART TX
2022-02-16_11_25_53_280: EngineState: TunerStudio size=640 / total=5005 / errors=0 / H=4 / O=4725 / P=83 / B=0
2022-02-16_11_25_53_280: EngineState: TunerStudio W=0 / C=0 / P=1
2022-02-16_11_25_53_281: EngineState: READ mode=1 offset=0
2022-02-16_11_25_53_355: EngineState: 0 - unix time
2022-02-16_11_25_53_356: EngineState: Current RTC localtime is: 2022-02-16 11:25:52 w=1
2022-02-16_11_25_53_357: EngineState: confirmation_set date 1645010752:19

warning: flash CRC failed, i never noticed this warning in console, i know in TS there were times when  had burn issues. I do not know if this is linked with the fix, but if i can reproduce it i will gather info and create a new ticket.

@mck1117 i would say this can be closed Thanks for the support

shadowm60, 2022-02-16:
played around in console and got this. is this reconnect fails? since console just shuts down, but from the log i do not understand what is the issue/cause

rlog Writing to logs/MAIN_rfi_report_2022-02-16_11_41_18_060.csv
I 220216 114118.077 [main] ConsoleUI - OS name: Windows 10
I 220216 114118.079 [main] ConsoleUI - OS version: 10.0
LazyFile: Not updating logs/README.html since looks to be the same content, new content size=171
Opening rusefi_tool.log
Console configuration from C:\Users\Alexa\.rusEFI\console_properties.xml
I 220216 114118.400 [AWT-EventQueue-0] MessagesCentral - postMessage Launcher: Available port: COM1
I 220216 114118.401 [AWT-EventQueue-0] MessagesCentral - postMessage Launcher: Available port: COM18
I 220216 114118.402 [AWT-EventQueue-0] MessagesCentral - postMessage Launcher: Available port: COM19
I 220216 114118.403 [AWT-EventQueue-0] MessagesCentral - postMessage Launcher: Available port: COM3
I 220216 114118.403 [AWT-EventQueue-0] MessagesCentral - postMessage Launcher: Available port: COM4
I 220216 114118.732 [AWT-EventQueue-0] Launcher - Rendering available ports: [Auto Serial, COM1, COM18, COM19, COM3, COM4]
I 220216 114120.160 [version checker] ConsoleUI - Reading from https://rusefi.com/console/versions.txt
I 220216 114120.164 [version checker] ConsoleUI - Server recommends java_console version 20171217 or newer
I 220216 114120.164 [version checker] ConsoleUI - Server recommends firmware 14892 or newer
I 220216 114120.621 [Ports Scanner] TcpConnector - Connection refused in getAvailablePorts(): simulator not running in 2028ms
I 220216 114120.794 [AWT-EventQueue-0] TcpConnector - Connection refused in getAvailablePorts(): simulator not running in 2031ms
communicationThread lookup DONE
class com.rusefi.io.CommandQueue: SerialIO started
2022-02-16_11_41_23_144<EOT>: Settings file not found or damaged. Using default settings.
I 220216 114123.426 [AWT-EventQueue-0] ConsoleUI - Console 20220215
I 220216 114123.427 [AWT-EventQueue-0] ConsoleUI - Hardware: F4
I 220216 114123.427 [AWT-EventQueue-0] LinkManager - LinkManager: Starting COM4
Zoom setValue 1.0
I 220216 114123.708 [Ports Scanner] TcpConnector - Connection refused in getAvailablePorts(): simulator not running in 2034ms
ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
line 8:6 missing ';' at '['
I 220216 114128.944 [AWT-EventQueue-0] MessagesCentral - postMessage ConsoleUI: COMPOSITE_OFF_RPM=300
I 220216 114128.946 [AWT-EventQueue-0] AuthTokenPanel - Auth token from settings: 7870c9c2-2af6-423d-ae3c-54b18324c052
Saved settings to C:\Users\Alexa\.rusEFI\console_properties.xml
I 220216 114129.769 [AWT-EventQueue-0] Launcher - onWindowOpened
I 220216 114129.772 [AWT-EventQueue-0] StreamConnector - StreamConnector: connecting
I 220216 114129.772 [AWT-EventQueue-0] StreamConnector - scheduleOpening
I 220216 114129.773 [communication executor1] StreamConnector - scheduleOpening>openPort
class com.rusefi.io.LinkManager$5: Opening port: COM4
I 220216 114129.776 [communication executor1] SerialIoStream - Using com.fazecast.jSerialComm 2.5.3
I 220216 114129.777 [communication executor1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort COM4
I 220216 114129.829 [communication executor1] BinaryProtocol - Got rusEFI 2022.02.16.mre_f7.1768592053  signature
I 220216 114129.834 [communication executor1] BinaryProtocol - current_configuration.rusefi_binary Local cache CRC 630a3896

I 220216 114129.836 [communication executor1] BinaryProtocol - rusEFI says tune CRC32 0x630a3896 1661614230

I 220216 114129.836 [communication executor1] BinaryProtocol - rusEFI says tune CRC16 0x3896 14486

I 220216 114129.837 [communication executor1] BinaryProtocol - Got configuration from controller.
I 220216 114129.884 [Commands Queue] BinaryProtocol - Sending [set date 1645011689]
I 220216 114129.891 [communication executor1] IniFileModel - Searching for rusefi*.ini in ..
I 220216 114129.892 [communication executor1] IniFileModel - Reading ..\rusefi_mre_f7.ini
I 220216 114129.969 [communication executor1] IniFileModel - Section [SettingGroups]
I 220216 114129.970 [communication executor1] IniFileModel - Section [MegaTune]
I 220216 114129.970 [communication executor1] IniFileModel - Section [TunerStudio]
I 220216 114129.971 [communication executor1] IniFileModel - Section [Constants]
I 220216 114130.024 [communication executor1] IniFileModel - Section [Tuning]
I 220216 114130.024 [communication executor1] IniFileModel - Section [LoggerDefinition]
I 220216 114130.028 [communication executor1] IniFileModel - Section [VeAnalyze]
I 220216 114130.036 [communication executor1] IniFileModel - Section [OutputChannels]
I 220216 114130.038 [communication executor1] IniFileModel - Section [PcVariables]
I 220216 114130.038 [communication executor1] IniFileModel - Section [ConstantsExtensions]
I 220216 114130.041 [communication executor1] IniFileModel - Section [CurveEditor]
I 220216 114130.042 [communication executor1] IniFileModel - Section [TableEditor]
I 220216 114130.043 [communication executor1] IniFileModel - Section [GaugeConfigurations]
I 220216 114130.044 [communication executor1] IniFileModel - Section [WueAnalyze]
I 220216 114130.049 [communication executor1] IniFileModel - Section [FrontPage]
I 220216 114130.056 [communication executor1] IniFileModel - Section [KeyActions]
[... cut ...]
Shutdown hook!  at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)

        at java.lang.Thread.run(Unknown Source)
I 220216 114145.030 [communication executor2] StreamConnector - scheduleOpening>openPort
class com.rusefi.io.LinkManager$5: Opening port: COM4
java.lang.IllegalStateException: Shutdown in progress
        at java.lang.ApplicationShutdownHooks.add(Unknown Source)class com.rusefi.io.LinkManager$5: Opening port: COM4
C:\Users\Alexa\Documents\TunerStudioProjects\c20xe\snapshot_20220216_063719_mre_f7_rusefi\console>

shadowm60, 2022-02-16:
runing engine simulator, an just looking at engine sniffer, console stays on and works fine, then check live data and this came up, now it is just frozen.
I 220216 115158.834 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: no CAM signals
I 220216 115210.900 [communication executor1] EngineState - EngineState.unknown: t1!u_50!0!t1!d_50!0!wave_chart`map!d!25!i1!u!33!i2!u!34!i3!u!34!i4!u!34!t1!u_52!83!t1!d_52!83!c3!d!146!t1!u_54!166!t1!d_54!166!t1!u_56!251!t1!d_56!251!t1!u_58!334!t1!d_58!334!t1!u_60!415!t1!d_60!416!i1!d!417!i2!d!418!i3!d!418!i4!d!418!t1!u_62!499!t1!d_62!499!t1!u_64!583!t1!d_64!584!t1!u_66!667!t1!d_66!667!t1!u_68!750!t1!d_68!750!t1!u_70!833!t1!d_70!833!t1!u_72!916!t1!d_72!917!t1!u_74!999!t1!d_74!1000!t1!u_76!1083!t1!d_76!1083!t1!u_78!1167!t1!d_78!1168!t1!u_80!1251!t1!d_80!1251!t1!u_82!1332!t1!d_82!1333!t1!u_84!1415!t1!d_84!1416!t1!u_86!1500!t1!d_86!1501!t1!u_88!1583!t1!d_88!1584!t1!u_90!1667!t1!d_90!1667!t1!u_92!1750!t1!d_92!1750!map!u!1830!t1!u_94!1833!t1!d_94!1833!t1!u_96!1916!t1!d_96!1917!t1!u_98!1999!t1!d_98!2000!t1!u_100!2083!t1!d_100!2083!t1!u_102!2166!t1!d_102!2166!c4!u!2247!t1!u_104!2249!t1!d_104!2249!t1!u_106!2332!t1!d_106!2333!t1!u_108!2417!t1!d_108!2417!t1!u_110!2500!t1!d_110!2501!map!d!2525!i1!u!2534!i2!u!2534!i3!u!2535!i4!u!2535!t1!u_112!2583!t1!d_112!2584!c4!d!2647!t1!u_114!2667!t1!d_114!2667!t1!u_116!2917!t1!d_116!2917!i1!d!2918!i2!d!2919!i3!d!2919!i4!d!2919!t1!u_118!2999!t1!d_118!3000!t1!u_120!3083!t1!d_120!3083!t1!u_122!3166!t1!d_122!3166!t1!u_124!3249!t1!d_124!3249!t1!u_126!3332!t1!d_126!3333!t1!u_128!3417!t1!d_128!3417!t1!u_130!3500!t1!d_130!3501!t1!u_132!3583!t1!d_132!3584!t1!u_134!3667!t1!d_134!3667!t1!u_136!3750!t1!d_136!3750!t1!u_138!3833!t1!d_138!3833!t1!u_140!3916!t1!d_140!3917!t1!u_142!4000!t1!d_142!4000!t1!u_144!4083!t1!d_144!4083!t1!u_146!4166!t1!d_146!4166!t1!u_148!4249!t1!d_148!4249!map!u!4330!t1!u_150!4334!t1!d_150!4334!t1!u_152!4417!t1!d_152!4417!t1!u_154!4499!t1!d_154!4499!t1!u_156!4582!t1!d_156!4582!t1!u_158!4667!t1!d_158!4667!c2!u!4747!t1!u_160!4750!t1!d_160!4750!t1!u_162!4833!t1!d_162!4833!t1!u_164!4916!t1!d_164!4917!t1!u_166!4999!t1!d_166!5000!map!d!5025!i1!u!5033!i2!u!5034!i3!u!5034!i4!u!5034!t1!u_168!5083!t1!d_168!5083!c2!d!5146!t1!u_170!5166!t1!d_170!5166!t1!u_172!5251!t1!d_172!5251!t1!u_174!5334!t1!d_174!5334!t1!u_176!5415!t1!d_176!5416!i1!d!5417!i2!d!5418!i3!d!5418!i4!d!5418!t1!u_178!5499!t1!d_178!5499!t1!u_180!5583!t1!d_180!5584!t1!u_182!5667!t1!d_182!5667!t1!u_184!5750!t1!d_184!5750!t1!u_186!5833!t1!d_186!5833!t1!u_188!5916!t1!d_188!5917!t1!u_190!6000!t1!d_190!6000!t1!u_192!6083!t1!d_192!6083!t1!u_194!6167!t1!d_194!6168!t1!u_196!6251!t1!d_196!6251!t1!u_198!6332!t1!d_198!6333!t1!u_200!6415!t1!d_200!6416!t1!u_202!6500!t1!d_202!6501!t1!u_204!6583!t1!d_204!6584!t1!u_206!6667!t1!d_206!6667!t1!u_208!6750!t1!d_208!6750!map!u!6830!t1!u_210!6833!t1!d_210!6833!t1!u_212!6916!t1!d_212!6917!t1!u_214!6999!t1!d_214!7000!t1!u_216!7083!t1!d_216!7083!t1!u_218!7166!t1!d_218!7166!c1!u!7247!t1!u_220!7249!t1!d_220!7249!t1!u_222!7332!t1!d_222!7333!t1!u_224!7417!t1!d_224!7417!t1!u_226!7500!t1!d_226!7501!map!d!7525!i1!u!7534!i2!u!7535!i3!u!7535!i4!u!7535!t1!u_228!7583!t1!d_228!7584!c1!d!7647!t1!u_230!7667!t1!d_230!7667!t1!u_0!7916!t1!d_0!7917!r!1200!7920!i1!d!7920!i2!d!7920!i3!d!7921!i4!d!7921!t1!u_2!7999!t1!d_2!8000!t1!u_4!8083!t1!d_4!8083!t1!u_6!8166!t1!d_6!8166!t1!u_8!8249!t1!d_8!8249!t1!u_10!8332!t1!d_10!8333!t1!u_12!8417!t1!d_12!8417!t1!u_14!8500!t1!d_14!8501!t1!u_16!8583!t1!d_16!8584!t1!u_18!8667!t1!d_18!8667!t1!u_20!8750!t1!d_20!8750!t1!u_22!8833!t1!d_22!8833!t1!u_24!8916!t1!d_24!8917!t1!u_26!8999!t1!d_26!9000!t1!u_28!9083!t1!d_28!9083!t1!u_30!9166!t1!d_30!9166!t1!u_32!9249!t1!d_32!9249!map!u!9330!t1!u_34!9334!t1!d_34!9334!t1!u_36!9417!t1!d_36!9417!t1!u_38!9499!t1!d_38!9499!t1!u_40!9582!t1!d_40!9582!t1!u_42!9667!t1!d_42!9667!c3!u!9747!t1!u_44!9750!t1!d_44!9750!t1!u_46!9833!t1!d_46!9833!t1!u_48!9916!t1!d_48!9917!t1!u_50!9999!t1!d_50!10000!map!d!10025!i1!u!10033!i2!u!10034!i3!u!10034!i4!u!10034!t1!u_52!10083!t1!d_52!10083!c3!d!10146!t1!u_54!10166!t1!d_54!10166!t1!u_56!10251!t1!d_56!10251!t1!u_58!10334!t1!d_58!10334!t1!u_60!10415!t1!d_60!10416!i1!d!10417!`
I 220216 115210.900 [communication executor1] EngineState - Invalid key [t1!u_50!0!t1!d_50!0!wave_chart] value [`map!d!25!i1!u!33!i2!u!34!i3!u!34!i4!u!34!t1!u_52!83!t1!d_52!83!c3!d!146!t1!u_54!166!t1!d_54!166!t1!u_56!251!t1!d_56!251!t1!u_58!334!t1!d_58!334!t1!u_60!415!t1!d_60!416!i1!d!417!i2!d!418!i3!d!418!i4!d!418!t1!u_62!499!t1!d_62!499!t1!u_64!583!t1!d_64!584!t1!u_66!667!t1!d_66!667!t1!u_68!750!t1!d_68!750!t1!u_70!833!t1!d_70!833!t1!u_72!916!t1!d_72!917!t1!u_74!999!t1!d_74!1000!t1!u_76!1083!t1!d_76!1083!t1!u_78!1167!t1!d_78!1168!t1!u_80!1251!t1!d_80!1251!t1!u_82!1332!t1!d_82!1333!t1!u_84!1415!t1!d_84!1416!t1!u_86!1500!t1!d_86!1501!t1!u_88!1583!t1!d_88!1584!t1!u_90!1667!t1!d_90!1667!t1!u_92!1750!t1!d_92!1750!map!u!1830!t1!u_94!1833!t1!d_94!1833!t1!u_96!1916!t1!d_96!1917!t1!u_98!1999!t1!d_98!2000!t1!u_100!2083!t1!d_100!2083!t1!u_102!2166!t1!d_102!2166!c4!u!2247!t1!u_104!2249!t1!d_104!2249!t1!u_106!2332!t1!d_106!2333!t1!u_108!2417!t1!d_108!2417!t1!u_110!2500!t1!d_110!2501!map!d!2525!i1!u!2534!i2!u!2534!i3!u!2535!i4!u!2535!t1!u_112!2583!t1!d_112!2584!c4!d!2647!t1!u_114!2667!t1!d_114!2667!t1!u_116!2917!t1!d_116!2917!i1!d!2918!i2!d!2919!i3!d!2919!i4!d!2919!t1!u_118!2999!t1!d_118!3000!t1!u_120!3083!t1!d_120!3083!t1!u_122!3166!t1!d_122!3166!t1!u_124!3249!t1!d_124!3249!t1!u_126!3332!t1! [... cut ...]
I 220216 115224.395 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: no CAM signals
I 220216 115237.198 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: no CAM signals
Selected paneNo: 8
I 220216 115253.271 [communication executor1] IncomingDataBuffer - get LiveDoc header: timeout 5000ms. Got only 0 while expecting 2
I 220216 115253.360 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: no CAM signals
I 220216 115253.447 [communication executor1] Launcher - ConnectionWatchdog.reconnectTimer restarting: 1000

shadowm60, 2022-02-16:
i could only break console with live data, tested on w11 on a new laptop, and it works, only live data would break it

rusefillc, 2022-02-16:
@shadowm60 any extra details for live data issue? let's follow up on it here or in a new ticket depending on what exception message says if you have on

shadowm60, 2022-02-16:
I've sharde all the data i could save. Console just shuts down, no error message, all that i have posted is output for java -jar rusefi_console.jar because just running the exe crashes.

Test made on mre f7, only usb connected, and on live data tab switching between cpp's causes the console to freak out

rusefillc, 2022-02-16:
let's have ticket open until we get a chance to do screen sharing session

rusefillc, 2022-09-01:
@shadowm60 shall we give it another try with more fresh firmware?

rusefillc, 2024-09-11:
looks abandoned

closed 2024-09-11 by rusefillc