rusefi/rusefi #3923: H7 console scare

rusefillc, 2022-02-12:
F4 good
[image]

H7 less good
[image]

rusefillc, 2022-02-12:
F4 :(

[image]

rusefillc, 2022-02-12:
win10 gives me 950 seconds and counting

[image]

rusefillc, 2022-02-12:
@mck1117 has

[image]

rusefillc, 2022-02-12:
BIOS update did not help :(

[image]

rusefillc, 2022-02-12:
TS is also affected it just does reconnect way nicer

```
2/12/22 7:46:07 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 7:46:28 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 7:46:49 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 7:47:10 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
```

TunerStudio uses different java connector "jssc" java-simple-serial-connector but experiences same timeout with rusEFI H7 every 20 seconds

```
2/12/22 7:43:18 AM :Time: 4:57.011: SENT 11 bytes, 11 bytes
x00 x05 x4F x00 x00 x00 x01 x6B x86 x08 x13                           ..O....k...

2/12/22 7:43:21 AM :Debug: Comm Exception Caught: Timeout after:3001ms. requested timeout:3000, Expected bytes:3, bytes read:0
Time: 5:00.011: READ Timout after 0 bytes, Expected:3, actual read:0 Raw buffer
Feb 12, 2022 7:43:21 AM y.u run
WARNING: Comm Timeout Caught
U.b: Timeout after:3001ms. requested timeout:3000, Expected bytes:3, bytes read:0
	at y.t.a(Unknown Source)
	at F.J.b(Unknown Source)
	at F.J.a(Unknown Source)
	at F.J.b(Unknown Source)
	at F.J.z(Unknown Source)
	at y.t.e(Unknown Source)
	at y.u.run(Unknown Source)

2/12/22 7:43:21 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 7:43:21 AM :Debug: JSSC ControllerInterface Close Connection currentPort == null:false
2/12/22 7:43:21 AM :Info: Closing Port
2/12/22 7:43:21 AM :Info: Successfully Closed Port
```

rusefillc, 2022-02-12:
Not related to #858

rusefillc, 2022-02-12:
Win11 laptop H7 20 seconds magic period

```
2/12/22 9:25:55 AM :Debug: isBlank Time: 0 ms.
2/12/22 9:25:55 AM :Debug: DiffTime: 17 ms.
2/12/22 9:26:08 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:26:29 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:26:50 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:27:11 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:27:32 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:27:55 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:28:16 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:28:37 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:28:58 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:29:19 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:29:40 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:30:01 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:30:22 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:30:43 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:31:05 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:31:30 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:31:51 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:32:12 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:32:33 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:32:54 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:33:21 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:33:42 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:34:03 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:34:24 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:34:45 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:35:45 AM :Info: Re-establishing connection to: RS232: Port:COM16, Baud:115200
2/12/22 9:36:35 AM :Info: Time to save msq: 41.2662
2/12/22 9:36:35 AM :Info: closeApp Called
```

rusefillc, 2022-02-12:
W10 had a perfect F4 discovery run - same exact discovery board with same cables

```
2/12/22 9:32:38 AM :Debug: Received Instruction: Write Chunk, Page: 1
2/12/22 9:32:39 AM :Debug: No Remaining Queue Write instructions
2/12/22 9:32:39 AM :Debug: No Remaining Queue Write instructions
2/12/22 9:32:39 AM :Info: Time to save msq: 18.459
2/12/22 9:32:39 AM :Info: Difference Report enabling shared table Models
2/12/22 9:32:45 AM :Debug: Deactivated fast paint Main Dashboard
2/12/22 10:46:49 AM :Info: Time to save msq: 54.2754
2/12/22 10:46:49 AM :Info: closeApp Called
2/12/22 10:46:49 AM :Info: Slave Server Deactivated.
2/12/22 10:46:49 AM :Debug: TuneLog Editor set text
2/12/22 10:46:50 AM :Debug: Fill Constants: 58
2/12/22 10:46:50 AM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:engineType equal to the proposed UNUSED25
2/12/22 10:46:50 AM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:debugMode equal to the proposed Alternator PID
```

rusefillc, 2022-02-12:
H7 is stable from TS on Win10 desktop

```
2/12/22 10:49:26 AM :Debug: DiffTime: 18 ms.
2/12/22 10:49:26 AM :Info: Getting ECU name for signature: 'rusEFI 2022.02.11.proteus_h7.2184629533'
2/12/22 10:49:26 AM :Info: Looks like rusEFI
2/12/22 10:49:26 AM :Info: Difference Report disabling shared table Models
2/12/22 10:49:28 AM :Debug: Fill Constants: 42
2/12/22 10:49:28 AM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:engineType equal to the proposed PROTEUS_DEFAULTS
2/12/22 10:49:28 AM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:idleMode equal to the proposed Manual
2/12/22 10:49:28 AM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:debugMode equal to the proposed Alternator PID
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeMinRpmMinTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeMinRpmMaxTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeMaxRpmMinTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeMaxRpmMaxTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeAirCoefMin!  found in current configuration, Min tCharge Coeff. found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeAirCoefMax!  found in current configuration, Max tCharge Coeff. found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeAirFlowMax! kg/h found in current configuration, Max Air Flow (kg/h) found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeAirIncrLimit! deg/sec found in current configuration, tChange Increment Limit (deg/sec) found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for tChargeAirDecrLimit! deg/sec found in current configuration, tChange Decrement Limit (deg/sec) found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Warning: mainUnusedEnd array size does not match the currently loaded configuration,
	rescaled .msq values to match configuration.
2/12/22 10:49:28 AM :Debug: An .msq value was null for fsioFormulas2??? This shouldn't happen.
2/12/22 10:49:28 AM :Debug: An .msq value was null for fsioFormulas3??? This shouldn't happen.
2/12/22 10:49:28 AM :Debug: An .msq value was null for fsioFormulas4??? This shouldn't happen.
2/12/22 10:49:28 AM :Debug: An .msq value was null for fsioFormulas5??? This shouldn't happen.
2/12/22 10:49:28 AM :Debug: An .msq value was null for fsioFormulas6??? This shouldn't happen.
2/12/22 10:49:28 AM :Warning: MSQ Units Mismatch for lambdaTable! afr found in current configuration, deg found in MSQ, values were not converted to new units.
2/12/22 10:49:28 AM :Info: Skip Restore Point for 'proteus-f7' because nothing changed since: proteus-f7_2022-02-12_08.45.00.msq
2/12/22 10:49:29 AM :Debug: No Remaining Queue Write instructions
2/12/22 10:49:29 AM :Debug: No Remaining Queue Write instructions
2/12/22 10:49:29 AM :Info: Time to save msq: 7.9624
2/12/22 10:49:29 AM :Info: Difference Report enabling shared table Models
2/12/22 1:08:10 PM :Info: Slave Server Deactivated.
2/12/22 1:08:10 PM :Debug: TuneLog Editor set text
2/12/22 1:08:10 PM :Debug: Fill Constants: 48
2/12/22 1:08:10 PM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:engineType equal to the proposed PROTEUS_DEFAULTS
2/12/22 1:08:10 PM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:idleMode equal to the proposed Manual
2/12/22 1:08:10 PM :Warning: Parameter in .msq, but not valid for current firmware: No options found for Bit EcuParameter:debugMode equal to the proposed Alternator PID
2/12/22 1:08:10 PM :Warning: MSQ Units Mismatch for tChargeMinRpmMinTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 1:08:10 PM :Warning: MSQ Units Mismatch for tChargeMinRpmMaxTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 1:08:10 PM :Warning: MSQ Units Mismatch for tChargeMaxRpmMinTps!  found in current configuration, mult found in MSQ, values were not converted to new units.
2/12/22 1:08:10 PM :Warning: MSQ Units Mismatch for tChargeMaxRpmMaxTps!  found in current configuration, mult found in MSQ, values were not converted to new units.

```

rusefillc, 2022-02-12:
Win10 desktop has issue with H7 at least with textPull=true

```
Sat Feb 12 13:09:56 EST 2022: onDataArrival alive for 19 second(s) textPull=true rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:09:56 EST 2022: onDataArrival alive for 19 second(s) textPull=true rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:09:56 EST 2022: onDataArrival alive for 19 second(s) textPull=true rusEFI 2022.02.11.proteus_h7.2184629533 
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: rusEfiVersion`20220211@36620 proteus MINIMAL_PINS 1431655765`outpin`t1@Digital 1`outpin`c1@Ign 1`outpin`i1@Lowside 1`outpin`c2@Ign 2`outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [rusEfiVersion] value [`20220211@36620 proteus MINIMAL_PINS 1431655765]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`t1@Digital 1`outpin`c1@Ign 1`outpin`i1@Lowside 1`outpin`c2@Ign 2`outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`t1@Digital 1]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`c1@Ign 1`outpin`i1@Lowside 1`outpin`c2@Ign 2`outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`c1@Ign 1]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`i1@Lowside 1`outpin`c2@Ign 2`outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`i1@Lowside 1]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`c2@Ign 2`outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`c2@Ign 2]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`i2@Lowside 2`outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`i2@Lowside 2]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`c3@Ign 3`outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`c3@Ign 3]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`i3@Lowside 3`outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`i3@Lowside 3]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`c4@Ign 4`outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`c4@Ign 4]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`i4@Lowside 4`outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`i4@Lowside 4]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`c5@Ign 5`outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`c5@Ign 5]
I 220212 130956.473 [communication executor1] EngineState - EngineState.unknown: outpin`i5@Lowside 5`outpin`c6@Ign 6`outpin`i6@Lowside 6`outpin`c7@Ign 7`outpin`i7@Lowside 7`outpin`c8@Ign 8`outpin`i8@Lowside 8`
I 220212 130956.473 [communication executor1] EngineState - Invalid key [outpin] value [`i5@Lowside 5]
[... cut ...]
E 220212 130956.580 [communication executor1] BinaryProtocol - output channels: executeCommand failed: java.io.IOException: write failed: wrote 0 but expected 11
writeFooter logs/rusEFI_trigger_log_2022-02-12_13_09_37_700.logicdata
I 220212 130956.584 [communication executor1] SerialIoStream - COM91: Closing port...
I 220212 130956.584 [communication executor1] SerialIoStream - COM91: Closed port.
I 220212 130956.691 [text pull1] BinaryProtocol - Stopping text pull

Process finished with exit code 130
```

rusefillc, 2022-02-12:
win10 H7 desktop issue textPull=false

```
I 220212 131533.635 [communication executor1] BinaryProtocol - Got rusEFI 2022.02.11.proteus_h7.2184629533  signature
I 220212 131533.641 [communication executor1] BinaryProtocol - current_configuration.rusefi_binary Local cache CRC 737e65ec

I 220212 131533.643 [communication executor1] BinaryProtocol - rusEFI says tune CRC32 0x737e65ec 1937663468

I 220212 131533.643 [communication executor1] BinaryProtocol - rusEFI says tune CRC16 0x65ec 26092

I 220212 131533.644 [communication executor1] BinaryProtocol - Got configuration from controller.
Sat Feb 12 13:15:33 EST 2022: onDataArrival alive for 0 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:33 EST 2022: onDataArrival alive for 0 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:33 EST 2022: onDataArrival alive for 1 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:33 EST 2022: onDataArrival alive for 1 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
I 220212 131538.999 [communication executor1] IncomingDataBuffer - composite log header: timeout 5000ms. Got only 0 while expecting 2
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 6 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:39 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 7 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:40 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 8 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Sat Feb 12 13:15:41 EST 2022: onDataArrival alive for 9 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
I 220212 131546.902 [communication executor1] IncomingDataBuffer - composite log header: timeout 5000ms. Got only 0 while expecting 2
I 220212 131551.914 [communication executor1] IncomingDataBuffer - output channels header: timeout 5000ms. Got only 0 while expecting 2
writeFooter logs/rusEFI_trigger_log_2022-02-12_13_15_33_663.logicdata
```

rusefillc, 2022-02-13:
3rd Windows device in test" win7 laptop. few hours of reliable MRE connection

```
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14732 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14732 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14732 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14733 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14733 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
Sat Feb 12 21:55:13 EST 2022: onDataArrival alive for 14733 second(s) textPull=false rusEFI 2022.02.12.mre_f4.1768592053 
```

rusefillc, 2022-02-13:
Some ideas to switch to linux at https://forum.chibios.org/viewtopic.php?f=16&t=6029

Also cool Luna device in a year from now

rusefillc, 2022-02-13:
for instance MRE F4
```
[Port2]  :  USB Composite Device

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            19
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&384dd860&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "rusEFI Engine Management ECU"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x11
Open Pipes:                           5

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0110
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x0483 = STMicroelectronics
idProduct:                       0x5740
bcdDevice:                       0x0200
iManufacturer:                     0x01
     English (United States)  "rusEFI LLC"
iProduct:                          0x02
     English (United States)  "rusEFI Engine Management ECU"
iSerialNumber:                     0x03
     English (United States)  "210033000A5056524D303220"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0xFF

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
[... 110 lines cut ...]
bInterfaceNumber:                  0x02
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0x0A  -> This is a CDC Data USB Device Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

```

rusefillc, 2022-02-13:
proteus H7

[image]

```
[Port2]  :  USB Composite Device

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            19
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&384dd860&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "rusEFI Engine Management ECU"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x12
Open Pipes:                           5

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0110
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x0483 = STMicroelectronics
idProduct:                       0x5740
bcdDevice:                       0x0200
iManufacturer:                     0x01
     English (United States)  "rusEFI LLC"
iProduct:                          0x02
     English (United States)  "rusEFI Engine Management ECU"
iSerialNumber:                     0x03
     English (United States)  "320037000751303434373339"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0008 = 0x08 bytes
bInterval:                         0xFF

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00
[... 113 lines cut ...]
bInterfaceNumber:                  0x02
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0x0A  -> This is a CDC Data USB Device Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

```

rusefillc, 2022-02-13:
We are definitely not honest about ``Self Powered``

thirstyone, 2022-02-13:
self powered from usb? were all these tests with the same usb cable?

rusefillc, 2022-02-13:
yes, yes.

thirstyone, 2022-02-13:
(Recently i was playing with a esp8266 board with a ch340 usb-serial converter on it. I bumped into random usb disconnect issues, which turned out to be due to voltage drop across the cable. All the problems had gone after I replaced the cable)

dron0gus, 2022-02-13:
Powered from USB is named "bus-powered"
https://github.com/rusefi/rusefi/pull/3933

mck1117, 2022-02-13:
Let me do some research about event providers to look at in Windows that may provide insight about what USB is doing.

mck1117, 2022-02-13:
> We are definitely not honest about `Self Powered`

H7 uses more power than F7 uses more power than F4 so this is actually a decent idea.

rusefillc, 2022-02-16:
on H7 sub-drama

[image]

rusefillc, 2022-02-16:
Win11, older firmware

```
Tue Feb 15 23:42:40 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Tue Feb 15 23:42:40 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Tue Feb 15 23:42:40 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Tue Feb 15 23:42:40 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Tue Feb 15 23:42:40 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
Tue Feb 15 23:42:41 EST 2022: onDataArrival alive for 38 second(s) textPull=false rusEFI 2022.02.11.proteus_h7.2184629533 
E 220215 234241.151 [communication executor1] BinaryProtocol - output channels: executeCommand failed: java.io.IOException: write failed: wrote -1 but expected 11
writeFooter logs/rusEFI_trigger_log_2022-02-15_23_42_03_868.logicdata
I 220215 234241.153 [communication executor1] SerialIoStream - COM16: Closing port...
I 220215 234241.153 [communication executor1] SerialIoStream - COM16: Closed port.
I 220215 234241.251 [text pull1] BinaryProtocol - Stopping text pull
```

rusefillc, 2022-02-16:
Win11, latest firmware :)
```
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 169 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 169 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 169 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 169 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 170 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
Tue Feb 15 23:48:05 EST 2022: onDataArrival alive for 170 second(s) textPull=false rusEFI 2022.02.16.proteus_h7.2184629533 
```

rusefillc, 2022-02-16:
Shit it works on Win11 with release?!

```
Tue Feb 15 23:55:00 EST 2022: onDataArrival alive for 88 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Tue Feb 15 23:55:00 EST 2022: onDataArrival alive for 88 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Tue Feb 15 23:55:00 EST 2022: onDataArrival alive for 88 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Tue Feb 15 23:55:00 EST 2022: onDataArrival alive for 88 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
```

https://github.com/rusefi/rusefi/actions/runs/1832661012 let me try https://github.com/rusefi/rusefi/suites/5275955657/artifacts/163192526

rusefillc, 2022-02-16:
well, time it's 74 seconds. Maybe a better test would be 300 seconds

```
Wed Feb 16 00:16:59 EST 2022: onDataArrival alive for 74 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:16:59 EST 2022: onDataArrival alive for 74 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:16:59 EST 2022: onDataArrival alive for 74 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:16:59 EST 2022: onDataArrival alive for 74 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:16:59 EST 2022: onDataArrival alive for 74 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
E 220216 001659.734 [communication executor1] BinaryProtocol - output channels: executeCommand failed: java.io.IOException: write failed: wrote -1 but expected 11
writeFooter logs/rusEFI_trigger_log_2022-02-16_00_16_46_255.logicdata
I 220216 001659.738 [communication executor1] SerialIoStream - COM16: Closing port...
I 220216 001659.738 [communication executor1] SerialIoStream - COM16: Closed port.
I 220216 001659.847 [text pull1] BinaryProtocol - Stopping text pull
```

rusefillc, 2022-02-16:
win11+h7
```
Wed Feb 16 00:46:59 EST 2022: onDataArrival alive for 303 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Wed Feb 16 00:46:59 EST 2022: onDataArrival alive for 303 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Wed Feb 16 00:46:59 EST 2022: onDataArrival alive for 304 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Wed Feb 16 00:46:59 EST 2022: onDataArrival alive for 304 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 
Wed Feb 16 00:47:00 EST 2022: onDataArrival alive for 304 second(s) textPull=false rusEFI 2022.01.26.proteus_h7.2020502113 

```

rusefillc, 2022-02-16:
win+h7 it makes no sense :(

```
Wed Feb 16 00:56:07 EST 2022: onDataArrival alive for 304 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:56:07 EST 2022: onDataArrival alive for 304 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:56:07 EST 2022: onDataArrival alive for 305 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
Wed Feb 16 00:56:07 EST 2022: onDataArrival alive for 305 second(s) textPull=false rusEFI 2022.02.12.proteus_h7.2184629533 
```