rusefi/rusefi #3935: console reconnection deadlock

mck1117, 2022-02-13:
[image]

Timer calls `LinkManager.restart()`, calls `connect()`, which calls `startAndConnect` etc which queues a function on the executor which will do the connection.

Then it calls `await` on a `CountDownLatch`, blocking the thread. But we're running on the executor thread! This blocks the executor from running the task that will complete, deadlocking the app.

mck1117, 2022-02-16:
Looks fixed to me, @rusefillc does the console survive if you push/release reset button 10 times?

rusefillc, 2022-02-18:
@mck1117 looks great!!!

@andreika-git FYI i know you were interested in that overall area

closed 2022-02-18 by rusefillc