rusefi/rusefi #3937: Dron reports that MRE console does not work from linux

rusefillc, 2022-02-14:
@dron0gus says https://rusefi.com/build_server/rusefi_bundle_mre_f4.zip does not connect to MRE from under linux

an easy way for me to test would be to use command line headless tools from the same jar

rusefillc, 2022-02-15:
I can confirm the issue but I do not have dev environment set on linux :( I am failing to even get terminal talking to RE on debian

```

debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$ lsusb
Bus 002 Device 012: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$ bin/detect.sh
I 220214 205503.298 [main] Launcher - rusEFI UI console 20220213
I 220214 205503.362 [main] Launcher - Compiled Mon Feb 14 12:49:06 EST 2022
I 220214 205503.362 [main] Launcher -

I 220214 205503.423 [main] PortDetector - Trying [ttyACM0]
I 220214 205503.429 [AutoDetectPort1] SerialIoStream - Using com.fazecast.jSerialComm 2.5.3
I 220214 205503.430 [AutoDetectPort1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
I 220214 205503.440 [AutoDetectPort1] SerialIoStream - ttyACM0: Closing port...
I 220214 205503.440 [AutoDetectPort1] SerialIoStream - ttyACM0: Closed port.
I 220214 205508.424 [main] PortDetector - Now interrupting [Thread[AutoDetectPort1,5,]]
I 220214 205508.424 [main] PortDetector - Interrupting Thread[AutoDetectPort1,5,]
I 220214 205508.425 [main] PortDetector - Done interrupting!
rusEFI not detected
Shutdown hook...
Shutdown hook!
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$ sudo cu -l /dev/ttyACM0
cu: open (/dev/ttyACM0): Permission denied
cu: /dev/ttyACM0: Line in use
debian@debian:~/snapshot_20220214_174906_mre_f4_rusefi$
```

rusefillc, 2022-02-21:
```
sudo java -jar console/rusefi_console.jar reboot_ecu

[sudo] password for debian:
I 220221 104531.089 [main] Launcher - rusEFI UI console 20220213
I 220221 104531.114 [main] Launcher - Compiled Mon Feb 14 12:49:06 EST 2022
I 220221 104531.114 [main] Launcher -

I 220221 104531.156 [main] PortDetector - Trying [ttyACM0, ttyS0]
I 220221 104531.162 [AutoDetectPort1] SerialIoStream - Using com.fazecast.jSerialComm 2.5.3
I 220221 104531.162 [AutoDetectPort1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
I 220221 104531.162 [AutoDetectPort2] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyS0
I 220221 104531.176 [AutoDetectPort1] SerialAutoChecker - Got signature=rusEFI 2022.02.20.mre_f4.1768592053 from ttyACM0
I 220221 104531.176 [AutoDetectPort1] SerialIoStream - ttyACM0: Closing port...
I 220221 104532.186 [AutoDetectPort1] SerialIoStream - ttyACM0: Closed port.
I 220221 104532.187 [AutoDetectPort1] SerialAutoChecker - Propagating AutoDetectResult{serialPort='ttyACM0', signature='rusEFI 2022.02.20.mre_f4.1768592053'}
I 220221 104532.187 [main] PortDetector - Now interrupting [Thread[AutoDetectPort1,5,], Thread[AutoDetectPort2,5,main]]
I 220221 104532.187 [main] PortDetector - Interrupting Thread[AutoDetectPort1,5,]
I 220221 104532.187 [main] PortDetector - Interrupting Thread[AutoDetectPort2,5,main]
I 220221 104532.187 [main] PortDetector - Done interrupting!
I 220221 104532.187 [AutoDetectPort2] SerialIoStream - ttyS0: Closing port...
I 220221 104532.189 [main] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
I 220221 104533.180 [AutoDetectPort2] SerialIoStream - ttyS0: Closed port.
Shutdown hook...
Shutdown hook!
```

rusefillc, 2022-02-21:
Maybe https://github.com/Fazecast/jSerialComm/issues/411 ?

rusefillc, 2022-02-22:
@dron0gus progress

```
debian@debian:~/snapshot_20220221_225035_mre_f4_rusefi$ bin/switch_to_dfu.sh
I 220221 200534.349 [main] Launcher - rusEFI UI console 20220221
I 220221 200534.375 [main] Launcher - Compiled Mon Feb 21 17:50:36 EST 2022
I 220221 200534.375 [main] Launcher -

I 220221 200534.416 [main] PortDetector - Trying [ttyACM0]
I 220221 200534.422 [AutoDetectPort1] SerialIoStream - Using com.fazecast.jSerialComm 2.7.0
I 220221 200534.422 [AutoDetectPort1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
E 220221 200534.423 [AutoDetectPort1] SerialIoStream - Error opening ttyACM0 maybe no permissions?
I 220221 200534.428 [AutoDetectPort1] SerialIoStream - ttyACM0: Closing port...
I 220221 200534.429 [AutoDetectPort1] SerialIoStream - ttyACM0: Closed port.
I 220221 200539.421 [main] PortDetector - Now interrupting [Thread[AutoDetectPort1,5,]]
I 220221 200539.421 [main] PortDetector - Interrupting Thread[AutoDetectPort1,5,]
I 220221 200539.421 [main] PortDetector - Done interrupting!
rusEFI not detected
Shutdown hook...
Shutdown hook!
```

billstewartx, 2022-02-26:
did you try a different port?
 "Trying [ttyACM0]" 
when you boot up Tunerstudio, does it connect okay? 
which port does Tunerstudio use ?

closed 2022-02-27 by rusefillc

rusefillc, 2023-03-08:
Works on WLS which is kind of Ubuntu

[image]

mi-hol, 2023-03-09:
> Works on WLS which is kind of Ubuntu

I ran it on WSL too but using Java 11

- OS version: 5.15.79.1-microsoft-standard-WSL2
- openjdk version "11.0.18" 2023-01-17

What java version did you use?

rusefillc, 2023-03-09:
[image]

dron0gus, 2023-03-09:
@mi-hol , console works just fine

```
$ java -version
openjdk version "1.8.0_362"
OpenJDK Runtime Environment (build 1.8.0_362-8u362-ga-0ubuntu1~18.04.1-b09)
OpenJDK 64-Bit Server VM (build 25.362-b09, mixed mode)
$ which java
/usr/bin/java
$ ls /usr/lib/jvm/java-11-openjdk-amd64/lib/libawt_xawt.so -lh
-rw-r--r-- 1 root root 389K Jan 20 12:07 /usr/lib/jvm/java-11-openjdk-amd64/lib/libawt_xawt.so
$ dpkg -S /usr/lib/jvm/java-11-openjdk-amd64/lib/libawt_xawt.so
openjdk-11-jre:amd64: /usr/lib/jvm/java-11-openjdk-amd64/lib/libawt_xawt.so
```

https://stackoverflow.com/questions/67823988/exception-in-thread-main-java-lang-unsatisfiedlinkerror-cant-load-library