rusefi/rusefi #3940: Extra Idle Air if Throttle is Pressed behavior

ElDominio, 2022-02-16:
I think this function should be slightly improved so it can better be used as a sort of dashpot.

Suggestions:
1) Make it engage when the ECU goes out of "idle" state, not when throttle is pressed
2)  Implement a check for RPM/s that when under a RPM/s threshold, the value is slowly (maybe the rate at which idle returns to normal can also be configured?) returned to normal idle values.

Example:
User pressed throttle, engine RPMs increase significantly
Once the RPMs start falling, check rate of change of RPM to make sure we aren't dropping like a stone and possibly shutting off (or still falling into steady idle)
Once RPM/s is relatively low, the additional idle air subtracts slowly, to be able to return to idle without an undershoot

[image]

rusefillc, 2022-04-03:
@andreika-git any interest?

mck1117, 2022-04-03:
@ElDominio do you have a log that demonstrates the behavior you want this change to avoid? That would be helpful at least for us to test against, and maybe to offer alternatives that are already possible today.

rusefillc, 2022-04-03:
and tune obviously

at the moment we are affected by https://github.com/rusefi/web_backend/issues/158 :(

ElDominio, 2022-04-04:
The behavior this would try to avoid, especially in Open Loop operation, is a dip of RPM as RPM comes down from not-idle. 

This post https://forums.linkecu.com/topic/13259-dash-pot-settings/?do=findComment&comment=85016 explains it quite well, and even some OEM's (basically all USDM) use this even in closed loop mode. You crack the idle valve open to a certain % as RPMs are dropping from above idle, wait for them to stabilize (for .5s-2s, depends on how hard said engine decelerates), then slowly return idle % to the base table. It allows for a much more mellow return to idle, and much more consistent in my experience.

mck1117, 2022-04-04:
Here's an example return to idle on rusEFI (5.3 LS, medium-ish cam, 228/230 duration):

[image]

My idle timing control settings are probably more aggressive than they need to be, but you get the idea.

BTW, you can emulate this behavior by setting your open loop idle position to be a little bit (50-100 RPM) higher than target, then let the closed loop control pull it down once you're close.

I'd still love to see a rusEFI log so we can compare apples-to-apples.

rusefillc, 2022-04-05:
[image]

LinkECU's "lock out" of idle control when going outside of idle parameters. It arms once the car goes out of idle control method (adds additional idle air %), then enters dashpot function once idle conditions have been met again, and does the behavior explained in the last image

[image]

additional idle % is held for x time, then removed over y amount of time to return to base idle value

rusefillc, 2022-04-05:
@ElDominio at the moment my plan is to catch up with @andreika-git tomorrow

but please consider sharing a relevant tune & log - we really need to know _what_ we are addressing not just _how_ we are addressing it.

ElDominio, 2022-04-05:
I want to share one, but I won't be near a RusEFI install in the next two weeks. I might be able to try one remotely, I'll let you guys know

ElDominio, 2022-04-19:
https://rusefi.com/online/view.php?log=750

This log shows the behavior

[image]

Blue circles are the problem (RPM dips below when coming to idle), and the green line I added would be a ROUGH estiamte of what i'd expect idle valve to do

rusefillc, 2022-04-20:
tune and log those come together as baskin and robbins

ElDominio, 2022-04-20:
It's linked in the post heh

rusefillc, 2022-04-20:
@ElDominio I must be blind :(

I do see https://rusefi.com/online/view.php?log=750 which is a log
I do not see tune link :(

mck1117, 2022-04-20:
[image]

mck1117, 2022-04-20:
@ElDominio have a look at lambda - it's suuuper lean on decel, which is causing the dip.  Stable idle is ~1000 rpm, ~33kPa, and it's only a few kPa low while on the decel, but the lambda sensor is indicating off scale lean (>1.35, which isn't much range).

[image]

rusefillc, 2022-04-20:
@mck1117 i do not trust that "auto-link tune" feature think I believe it's showing weird too often at least that's my experience maybe I am not nice to it :(

mck1117, 2022-04-20:
the linked one *is* the correct tune, though

mck1117, 2022-04-20:
> @mck1117 i do not trust that "auto-link tune" feature think I believe it's showing weird too often at least that's my experience maybe I am not nice to it :(

create ticket?

ElDominio, 2022-04-20:
Is it wrong to decel with no fuel? Shouldn't fueling be correct once the VE table goes round the idle areas?

rusefillc, 2022-04-20:
https://github.com/rusefi/web_backend/issues/169

mck1117, 2022-04-20:
> Is it wrong to decel with no fuel? Shouldn't fueling be correct once the VE table goes round the idle areas?

wall wetting means this isn't necessarily true

ElDominio, 2022-04-21:
While we could banter on and on about this.... Could it still be added? All I know is that I really like this function even if it's superflous/unnecessary, and it helps me a lot with customer cars.

rusefillc, 2022-04-21:
I am aware of three developers who could added
@mck1117 what do you say?
@andreika-git I suspect not sooner than a month from now
myself more or less same at least a month from now

ElDominio, 2022-10-31:
another explanation of what i'm looking for 
[image]

[image]

ElDominio, 2023-06-02:
gonna make new thread

closed 2023-06-02 by ElDominio