rusefi/rusefi #3942: ETB#1 & ETB#2 error

tmbryhn, 2022-02-18:
Playing with the analog inputs using a stimulator on my ECU running Proteus F4 FW.

With 2x ETB channels activated, the ECU will only stop injection and display a "tps Error" flag in TS if one of the two TPS-channels associated with ETB#1 is out of tolerance. I can play around with the TPS analog channels associated with ETB#2 and simulate an error without getting any reaction or error in TS.

I assume this error function and characteristic should be equal for both ETB channels.

mck1117, 2022-02-18:
That flag means "problem with this TPS sensor".  There's a corresponding indicator for TPS2 - it should go red when TPS2 is enabled (aka, at least one input channel is set) AND the sensor goes out of range or has a redundancy error.

tmbryhn, 2022-02-18:
Gotcha. I'll look for the indicator flag for TPS2.
But, the injection routine is halted for TPS1 error, but not for TPS2.

tmbryhn, 2022-02-18:
Yes. All injector outputs goes silent when TPS error flag is red.

mck1117, 2022-02-18:
Then something else is going on, we don't explicitly do something like "if tps fail, no injection". Can you upload your tune and a log of it happening to rusefi.com/online and link them both here?

tmbryhn, 2022-02-18:
Stimulator | Generic V12:

https://rusefi.com/online/?vehicleName=Stimulator&user_id=1083&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

Let me know if you need any other input.

tmbryhn, 2022-02-19:
Makes perfect sense.
While you mention it; is there any plan on implementing some form of synchronization between 2x TBs as a function of eg. MAP from two cylinder banks? I'm thinking along the line of a simple offset that can be applied manually in software to one of the TBs to balance the MAP signals.
Would've been useful for eg. ETB setup on boxer engines.

mck1117, 2022-03-19:
yes, there is an issue for that: #3680

closed 2022-03-19 by mck1117