rusefi/rusefi #3943: Onboard SD Card Datalogging Doesn't Record TPS

droneperson42, 2022-02-18:
After discovering the magic of SD card logs recently I found that it does not record "Throttle pedal position" or "TPS" parameters (TPS is the parameter used on my Hellen NB1 ECU). The TPS gauge in TS works fine and it appears in datalogs taken through TS with a laptop or shadowdash MS on my phone via BT. See the SD log [here](https://rusefi.com/online/view.php?log=656). Currently running December 29 2021 FW version

rusefillc, 2022-02-19:
code says that PPS is logged both under TPS and PPS names

```
	{engine->outputChannels.throttlePedalPosition, GAUGE_NAME_TPS, "%", 2},
	{engine->outputChannels.TPS2Value, GAUGE_NAME_TPS2, "%", 2},
	{engine->outputChannels.throttlePedalPosition, GAUGE_NAME_THROTTLE_PEDAL, "%", 2},
```

even your sample file shows TPS and PPS fields both at zero so on the one hand we have a defect but on the other hand it's not exactly correct to say that both TPS and PPS are not recorded. It looks like what's happening is PPS is recorded as TPS.

Do you have PPS on your vehicle? Please provide TS log along your next SD card log.

Now pushing fix to log TPS as TPS, firmware snapshot would have it in about 20 minutes

rusefillc, 2022-02-19:
@droneperson42 we definitely had at least one defect which is fixed now please test again when you have a chance

droneperson42, 2022-02-19:
I'll try it now. He's correct standard MX5 NB1 throttle cable and TPS

rusefillc, 2022-02-19:
``now`` would be too early. please give it another 11 minutes

droneperson42, 2022-02-19:
Yep, it fully works now! TPS values are displayed in the log. I've also noticed that the "date created" on the SD log files is also correct wheras before they all showed created in 2017. Cheers!

closed 2022-02-19 by droneperson42