rusefi/rusefi #3971: revisit romraider usage

rusefillc, 2022-03-01:
on the one hand RomRaider has made some amazing progress recently https://github.com/RomRaider/RomRaider/pull/130

on the other hand I think they are still officially 1.6 only and 32 bits only? that bother me a bit

For March 2022 we should remove RR under #319

Maybe one day we would add it again with proper upstream merge of all our integration hacks on top of 2021 refactoring progress

rusefillc, 2022-03-01:
FYI @racer-coder since you have interest in frontend side of things

mck1117, 2023-01-20:
looks like no romraider left in the repo?

[image]

closed 2023-01-21 by rusefillc