rusefi/rusefi #3980: ECU prints unknown command for all commands

mck1117, 2022-03-05:
ECU prints unknown command even when it's a known command

[image]

rusefillc, 2022-03-05:
@dron0gus could this be related to some of your recent changes?

dron0gus, 2022-03-05:
Yes. This is my fault. Nothing bad happens, only return code is incorrectly interpreted and error message emitted.
Will fix it on weekend.

mck1117, 2022-03-06:
Yep, that's why I assigned to you :)

No big deal, just created an issue so we didn't forget about it.

mck1117, 2022-03-06:
also let's not do this:

[image]

dron0gus, 2022-03-08:
https://github.com/rusefi/rusefi/pull/3987

closed 2022-03-11 by rusefillc