rusefi/rusefi #3986: LUA output & Injector 8 conflict cause no output to VR_THRESH

NMSTEC, 2022-03-08:
This is a place holder tonight so I don't forget to update with all info I find tomorrow.
[image]
[image]

NMSTEC, 2022-03-08:
Easy, this fatal error, was not there. It did not show up in tuner studio. It did not show up in console. We went through mutltiple reboots, we went through 2 proteuses. Finally the fatal occurred when without changing anything, I clicked write lua.

Good news, since this problem occurred on both ecus, I only made one work. The other still has this issue.

How would you like me to log it? Would you like to also see screen recording?

NMSTEC, 2022-03-08:
https://rusefi.com/online/view.php?msq=910

[image]
[image]
[image]

This is the second ecu that still has no output.
As you can clearly see there is no error, or fatal. If I click that write lua button, it will show fatal. This lua code is part of the tune, and was pulled by console from tune. I did not add it. LUA code IS functional as the bike will crank over. Bike will only crank over if there is CAN data being output.

NMSTEC, 2022-03-08:
`2022-03-08_08_18_19_819: ConsoleUI: COMPOSITE_OFF_RPM=300
2022-03-08_08_18_20_549: EngineState: initHardware()
2022-03-08_08_18_20_550: EngineState: led: CRITICAL status on PE3
2022-03-08_08_18_20_551: EngineState: initRtc()
2022-03-08_08_18_20_551: EngineState: engine chart: No
2022-03-08_08_18_20_552: EngineState: engine chart size=0
2022-03-08_08_18_20_552: EngineState: USB DM on PA11
2022-03-08_08_18_20_553: EngineState: USB DP on PA12
2022-03-08_08_18_20_553: EngineState: *** rusEFI LLC (c) 2012-2022. All rights reserved.
2022-03-08_08_18_20_554: EngineState: *** rusEFI v20220306@36981
2022-03-08_08_18_20_554: EngineState: *** Chibios Kernel:       6.1.4
2022-03-08_08_18_20_555: EngineState: *** Compiled:     Mar  6 2022 - 13:57:22
2022-03-08_08_18_20_556: EngineState: *** COMPILER=9.3.1 20200408 (release)
2022-03-08_08_18_20_556: EngineState: *** detected HSE clock 7.95 MHz PLLM = 8
2022-03-08_08_18_20_557: EngineState: UID=2B003F 34385115 30353633
2022-03-08_08_18_20_557: EngineState: can read 0x20000010 1
2022-03-08_08_18_20_558: EngineState: can read 0x20020010 1
2022-03-08_08_18_20_558: EngineState: can read 0x20070010 1
2022-03-08_08_18_20_559: EngineState: MCU rev=451 flashSize=1024
2022-03-08_08_18_20_559: EngineState: CORTEX_MAX_KERNEL_PRIORITY=2
2022-03-08_08_18_20_560: EngineState: STM32_ADCCLK=27000000
2022-03-08_08_18_20_560: EngineState: STM32_TIMCLK1=108000000
2022-03-08_08_18_20_561: EngineState: STM32_TIMCLK2=216000000
2022-03-08_08_18_20_561: EngineState: STM32_PCLK1=54000000
2022-03-08_08_18_20_562: EngineState: STM32_PCLK2=108000000
2022-03-08_08_18_20_562: EngineState: PORT_IDLE_THREAD_STACK_SIZE=32
2022-03-08_08_18_20_563: EngineState: CH_DBG_ENABLE_ASSERTS=1
2022-03-08_08_18_20_563: EngineState: CH_DBG_SYSTEM_STATE_CHECK=1
2022-03-08_08_18_20_564: EngineState: CH_DBG_ENABLE_STACK_CHECK=1
2022-03-08_08_18_20_564: EngineState: EFI_LOGIC_ANALYZER=0
2022-03-08_08_18_20_565: EngineState: EFI_TUNER_STUDIO=1
2022-03-08_08_18_20_565: EngineState: EFI_SIGNAL_EXECUTOR_SLEEP=0
2022-03-08_08_18_20_566: EngineState: EFI_SIGNAL_EXECUTOR_HW_TIMER=0
2022-03-08_08_18_20_566: EngineState: EFI_SHAFT_POSITION_INPUT=1
2022-03-08_08_18_20_567: EngineState: EFI_INTERNAL_ADC=1
2022-03-08_08_18_20_567: EngineState: readFromFlash 80C0000
2022-03-08_08_18_20_568: EngineState: applyNonPersistentConfiguration()
2022-03-08_08_18_20_568: EngineState: initializeTriggerWaveform(TT_TOOTHED_WHEEL_32_2/69)
2022-03-08_08_18_20_569: EngineState: initializeTriggerWaveform(TT_ONE/18)
2022-03-08_08_18_20_570: EngineState: Read valid configuration from flash!
2022-03-08_08_18_20_570: EngineState: led: comm status on PE4
2022-03-08_08_18_20_571: EngineState: led: warning status on PE6
2022-03-08_08_18_20_572: EngineState: led: running status on PE5
2022-03-08_08_18_20_572: EngineState: initAdcInputs()
2022-03-08_08_18_20_572: EngineState: knock ch1 on PF4
2022-03-08_08_18_20_573: EngineState: knock ch2 on PF5
2022-03-08_08_18_20_573: EngineState: SPI clock on PC10
2022-03-08_08_18_20_574: EngineState: SPI master out on PC12
2022-03-08_08_18_20_574: EngineState: SPI master in  on PC11
2022-03-08_08_18_20_575: EngineState: SD CS on PD2
2022-03-08_08_18_20_575: EngineState: Injector 1 on Lowside 1
2022-03-08_08_18_20_576: EngineState: Injector 2 on Lowside 2
2022-03-08_08_18_20_577: EngineState: Coil 1 on Ign 1
2022-03-08_08_18_20_577: EngineState: Coil 2 on Ign 2
2022-03-08_08_18_20_577: EngineState: Fan Relay on Lowside 15
2022-03-08_08_18_20_578: EngineState: extiTriggerTurnOnInputPin Trigger #1 Digital 6
2022-03-08_08_18_20_579: EngineState: Trigger #1 on Digital 6
2022-03-08_08_18_20_579: EngineState: extiTriggerTurnOnInputPin cam1 Digital 4
2022-03-08_08_18_20_580: EngineState: cam1 on Digital 4
2022-03-08_08_18_20_580: EngineState: CAN TX on PD1
2022-03-08_08_18_20_581: EngineState: CAN RX on PD0
2022-03-08_08_18_20_581: EngineState: CAN2 TX on PB13
2022-03-08_08_18_20_582: EngineState: CAN2 RX on PB12
2022-03-08_08_18_20_582: EngineState: initHardware() OK!
2022-03-08_08_18_20_582: EngineState: BatteryVoltage on PA7
2022-03-08_08_18_20_583: EngineState: MapSlow on Analog Volt 1
2022-03-08_08_18_20_583: EngineState: Tps1Primary on Analog Volt 2
2022-03-08_08_18_20_584: EngineState: Tps1Secondary on Analog Volt 3
2022-03-08_08_18_20_584: EngineState: AcceleratorPedalPrimary on Analog Volt 4
2022-03-08_08_18_20_585: EngineState: AcceleratorPedalSecondary on Analog Volt 9
[... 117 lines cut ...]
2022-03-08_08_18_20_637: EngineState: READ mode=1 offset=20736 size=256
2022-03-08_08_18_20_637: EngineState: READ mode=1 offset=20992 size=20
2022-03-08_08_18_20_639: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_639: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_639: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_640: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_640: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_641: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_18_20_641: EngineState: WARNING: ADC: ego i
2022-03-08_08_18_20_757: EngineState: 0 - unix time
2022-03-08_08_18_20_760: EngineState: Current RTC localtime is: 2022-03-08 08:18:20 w=1
2022-03-08_08_18_20_763: EngineState: unknown command [set date 1646727500]
2022-03-08_08_18_29_193: EngineState: WARNING: ADC: ego input is not configured
[... cut ...]
2022-03-08_08_19_19_281: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_19_29_216: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_19_39_259: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_19_49_194: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_19_59_213: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_20_09_248: EngineState: WARNING: ADC: ego input is not configured
2022-03-08_08_20_19_182: EngineState: WARNING: ADC: ego input is not configured
`

NMSTEC, 2022-03-08:
Doing the procedure above of resolving the pin collision, then rebooting and setting them again causes problem to go away, and everything works dandy. Looks like an edge case specifically with my lua code, and the order of actions I did. However, would still like to see if we can find root cause to make sure it doesn't occur with customer.

mck1117, 2022-04-03:
I think you changed the number of cylinders without rebooting the ECU, or something like that. This seems pretty unlikely for real customers.

closed 2022-04-03 by mck1117