rusefi/rusefi #3996: TPS AE engages during crank

NMSTEC, 2022-03-14:
Expected result: TPS vs TPS to not be active until engine is running.
Actual result: During cranking TPS will move to crank position, which may trip AE.

Conditions: ETB is disabled while engine not spinning.

I am unsure if cranking RPM is overriden by TPS acceleration or it stays static until out of crank condition however due to my engine type, we have long enrichment times, so it may roll over into normal VE table.

[image]
https://rusefi.com/online/view.php?log=694

rusefillc, 2022-03-14:
@NMSTEC see arguing at https://github.com/rusefi/rusefi/pull/3997

NMSTEC, 2022-03-14:
Argument continuing in Slack before transferring main talking points here.

NMSTEC, 2022-03-18:
Considering conversation did not continue, I say, TPS vs TPS AE should NOT be used during cranking or no spinning. If there needs to be enrichment by TPS during cranking, then either tune is bad, or you have a carb making EFI pointless here, or your engine is scrap.

commit 3bfe7f17 (2022-03-19): TPS AE engages during crank fix #3996 (#3997)
closed 2022-03-19 by rusefillc