rusefi/rusefi #4008: rusEFI console should not close on disconnect

rusefillc, 2022-03-18:
At the moment rusEFI console closes on disconnect from ECU - not ideal

mck1117, 2022-03-18:
Does not repro - I unplug ECU, it goes to "disconnected". Plug in, goes back to connected.

rusefillc, 2022-03-18:
please hit 'stm32_standby'

mck1117, 2022-03-18:
Works fine. Console goes to "disconnected". Press reset button on ECU. Console reconnects.

Tested with latest snapshot.

NMSTEC, 2022-03-20:
Mine disconnects and closes, within 10 seconds.

rusefillc, 2022-03-23:
unable to reproduce on Win11 with Proteus F4

rusefillc, 2022-03-23:
still an issue with win10 alphax-2

```
I 220323 144204.401 [communication executor1] MessagesCentral - postMessage EngineState: READ mode=1 offset=20224 size=256
I 220323 144204.401 [communication executor1] MessagesCentral - postMessage EngineState: READ mode=1 offset=20480 size=256
I 220323 144204.402 [communication executor1] MessagesCentral - postMessage EngineState: READ mode=1 offset=20736 size=256
I 220323 144204.402 [communication executor1] MessagesCentral - postMessage EngineState: READ mode=1 offset=20992 size=20
I 220323 144204.405 [Commands Queue] BinaryProtocol - Sending [set date 1648046524]
I 220323 144204.435 [communication executor1] MessagesCentral - postMessage EngineState: 0 - unix time
I 220323 144204.435 [communication executor1] MessagesCentral - postMessage EngineState: Current RTC localtime is: 2022-03-23 14:42:04 w=1
I 220323 144204.435 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_set date 1648046524:19
I 220323 144206.051 [Commands Queue] BinaryProtocol - Sending [stm32_standby]
I 220323 144208.008 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144209.021 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144210.036 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144211.040 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144211.104 [communication executor1] IncomingDataBuffer - execute header: timeout 5000ms. Got only 0 while expecting 2
E 220323 144211.104 [communication executor1] BinaryProtocol - output channels: executeCommand failed: java.io.IOException: write failed: wrote -1 but expected 11
I 220323 144211.110 [communication executor1] SerialIoStream - COM41: Closing port...
I 220323 144211.110 [communication executor1] SerialIoStream - COM41: Closed port.
I 220323 144211.110 [communication executor1] ConnectionWatchdog - ConnectionWatchdog.reconnectTimer restarting: 1000
I 220323 144211.139 [communication executor1] LinkManager - restart isPortAvailableAgain=true
I 220323 144211.140 [communication executor1] LinkManager - LinkManager: Starting COM41
I 220323 144211.140 [communication executor1] StreamConnector - StreamConnector: connecting
I 220323 144211.140 [communication executor1] StreamConnector - scheduleOpening
I 220323 144211.140 [communication executor1] StreamConnector - scheduleOpening>openPort
I 220323 144211.140 [communication executor1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort COM41
E 220323 144211.140 [communication executor1] SerialIoStream - Error opening COM41 maybe no permissions?
```

rusefillc, 2022-03-23:
[image]

rusefillc, 2022-03-23:
```
I 220323 144949.112 [AWT-EventQueue-0] BinaryProtocolServer - ServerSocket 2390 created
I 220323 144949.192 [Commands Queue] BinaryProtocol - Sending [set date 1648046989]
I 220323 144949.309 [communication executor1] MessagesCentral - postMessage EngineState: 0 - unix time
I 220323 144949.309 [communication executor1] MessagesCentral - postMessage EngineState: Current RTC localtime is: 2022-03-23 14:49:49 w=1
I 220323 144949.309 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_set date 1648046989:19
I 220323 144950.900 [Commands Queue] BinaryProtocol - Sending [stm32_standby]
I 220323 144952.897 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144953.903 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144954.909 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144955.912 [AWT-EventQueue-0] ConnectionWatchdog - restart already pending...
I 220323 144955.991 [communication executor1] IncomingDataBuffer - execute header: timeout 5000ms. Got only 0 while expecting 2
E 220323 144955.991 [communication executor1] BinaryProtocol - output channels: executeCommand failed: java.io.IOException: write failed: wrote -1 but expected 11
writeFooter logs/rusEFI_trigger_log_2022-03-23_14_49_48_652.logicdata
I 220323 144955.998 [communication executor1] SerialIoStream - COM41: Closing port...
I 220323 144955.998 [communication executor1] SerialIoStream - COM41: Closed port.
I 220323 144955.999 [communication executor1] ConnectionWatchdog - ConnectionWatchdog.reconnectTimer restarting: 1000
I 220323 144956.022 [communication executor1] LinkManager - restart isPortAvailableAgain=true
I 220323 144956.023 [communication executor1] LinkManager - LinkManager: Starting COM41
I 220323 144956.024 [communication executor1] StreamConnector - StreamConnector: connecting
I 220323 144956.024 [communication executor1] StreamConnector - scheduleOpening
I 220323 144956.024 [communication executor1] StreamConnector - scheduleOpening>openPort
class com.rusefi.io.LinkManager$5: Opening port: COM41
I 220323 144956.024 [communication executor1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort COM41
E 220323 144956.024 [communication executor1] SerialIoStream - Error opening COM41 maybe no permissions?
E 220323 144956.024 [communication executor1] IoUtils - CONNECTION FAILED, did you specify the right port name?
java.lang.Exception
	at com.rusefi.util.IoUtils.exit(IoUtils.java:16)
	at com.rusefi.io.LinkManager$3.onConnectionFailed(LinkManager.java:97)
	at com.rusefi.io.serial.PortHolder.connectAndReadConfiguration(PortHolder.java:59)
	at com.rusefi.io.serial.StreamConnector.lambda$connectAndReadConfiguration$0(StreamConnector.java:36)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Disconnected from the target VM, address: '127.0.0.1:58066', transport: 'socket'

```

rusefillc, 2022-03-23:
closing issue but we still have a few weird "system.exit" where we should have handled IO issues

fixing immediate issue and moving on

commit fbf58d70 (2022-03-23): rusEFI console should not close on disconnect fix #4008
closed 2022-03-23 by rusefillc