rusefi/rusefi #4009: AEM X-Series not reporting AFR's correctly over CAN BUS.

pk386, 2022-03-19:
[STM32F407 Discovery board on Frankenso)
running:  _snapshot_20220305_140027_stm32f407_rusefi_
Last known snapshot where lambda reported correctly: _snapshot_20210516_143957_stm32f407_rusefi_

- 30-0310 X-Series Inline Wideband UEGO Sensor Controller used
[30-0310 X-Series Inline Wideband UEGO Sensor Controller.pdf](https://github.com/rusefi/rusefi/files/8308394/30-0310.X-Series.Inline.Wideband.UEGO.Sensor.Controller.pdf)

CAN logs from the rusEFI console show data coming into the stm32 correctly. 
Output Lambda value in tuner studio is garbage.

**Is there anything I can check? 
Is there some variable that I can print to the console?**

I have attached an .ODS file that is a 2 logs from the AEM wide band (RS-232) port and the CAN logs from rusEFI combined.
[CANandRS232.ods](https://github.com/rusefi/rusefi/files/8308367/CANandRS232.ods)

Log file: 
https://rusefi.com/online/view.php?log=698

Tune:
https://rusefi.com/online/view.php?msq=918

White line is lambda value:
[image]

Decoder ring from AEM manual
[image]

Thank you for taking the time to report an issue!

To make our work easier, please provide as many details as you can:
* [ ] What hardware did you use while experiencing the issue?

pk386, 2022-04-02:
No change with _snapshot_20220401_173155_stm32f407_rusefi_

will upload log from console later.

pk386, 2022-04-02:
[Log from snapshot-2022-04-01.txt](https://github.com/rusefi/rusefi/files/8401695/Log.from.snapshot-2022-04-01.txt)

pk386, 2022-04-03:
Bad CAN bus transceiver on Frankenso board... 
replaced CAN bus x-ceiver. works now.
Key to finding the issue was looking at the ratio of canWrite___ and canRead___ variables.
[image]

closed 2022-04-03 by pk386

rusefillc, 2022-04-03:
this was a hardware issue, 5v vs 3v power/CAN chip

pk386, 2022-04-03:
> this was a hardware issue, 5v vs 3v power/CAN chip

Edit: the original 3.3V chip had issues, It was replaced with a 5v chip by error by the user.
Once disconnected from 3.3V and connected to 5V the replacement chip worked.