rusefi/rusefi #4011: Mitsubishi 3A92 Support

ElDominio, 2022-03-19:
I have a 3 cylinder Mitsubishi Mirage project car with a 3A92 engine, with this trigger wheel combo in the following pictures. I have some plans and would like trigger support to get it running ASAP!

Needs to be full sequential as this is a 3 cylinder lol
The trigger seems to be a 36-2-1-1, where the -2 indicates the arrival of piston https://github.com/noisymime/speeduino/issues/1 at one of its TDC's, while the cam wheel seems to be a 4+1, the the double tooth indicates that cylinder 1 is next at the true TDC.
[image]
[image]
[image]

rusefillc, 2022-03-19:
@ElDominio what years use this? is this ever VVT or always fixed position?

ElDominio, 2022-03-19:
This is always single VVT on exhaust, which is the wheel in the picture

rusefillc, 2022-03-20:
[image]

rusefillc, 2022-03-20:
@ElDominio https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#mitsubishi

rusefillc, 2022-03-20:
@ElDominio please test with Sunday firmware

closed 2022-03-20 by rusefillc