rusefi/rusefi #4012: Clarify trigger names to avoid confusion

ElDominio, 2022-03-19:
I think all "minus" triggers should be clarified, like 60-2 should be "60-2" not "60/2"  since the slash notation is what Megasquirt used (uses?) to refer to cam teeth. 36/2 in some ECUs would usually mean 36 on crank 2 on cam.

Also, there is a trigger called "Mitsubishi", but Mitsubishi uses a ton of pretty much custom trigger patters. 4G63 uses one pattern (well known, probably this is what "Mitsubishi" trigger is for), 6G71-4 use another pattern, 6G75 uses another pattern, 3A92 uses another pattern, 4B11 uses another pattern

ElDominio, 2022-12-13:
[image]

This is what I meant. Would help a lot to avoid confusion

ElDominio, 2024-02-23:
Still relevant,
there's 36/2, but then 36_2_2_2, then also 18-2-2-2

Why is it so incosistent? Why are slashes being used when no one uses slashes?

rusefillc, 2024-04-21:
Dup of #6258

closed 2024-04-21 by rusefillc