rusefi/rusefi #4014: Mitsubishi 6G75 Support

ElDominio, 2022-03-19:
[image]

Crank is a 36-2-1-1
Cam is that weird thing in the picture

rusefillc, 2022-03-19:
@ElDominio what years are those? does this ever have VVT or is this always fixed cam position?

ElDominio, 2022-03-19:
This has VVT, but it is only switched (like old VTEC)

rusefillc, 2022-03-20:
@ElDominio any chance of a straight image of the VVT trigger wheel?

ElDominio, 2022-04-08:
I can grab a scope off the oscilloscope but a picture of the cam itself.... let me try

rusefillc, 2022-09-28:
@ElDominio do we still care? if we do please provide more info :)

ElDominio, 2022-09-28:
This is a very common swap in PR, but as far as I can tell it's exactly 3A92. I haven't had one come in, but as soon as I do I will pester lol

rusefillc, 2022-09-28:
Let's reopen if we have anything to do

closed 2022-09-28 by rusefillc